Ysurac/openmptcprouter #2649: OpenMPTCPRouter in high latency environments

hkolvenbach, 2022-11-08:
I am currently in Peru which has terrible internet throughout (both landline and mobile) with very high latencies to the next cloud provider (i am using a GCP instance in chile) and get latencies between 60-200ms on landline and 200-1000ms over mobile.

Are there any configuration options to make OpenMPTCPRouter work ok under these circumstances? I often get messages on one of the links that the server didn't respond in 1 second. I am currently bundling 2 fixed wireless (which I think still uses LTE under the hood at some point) and one LTE connection, but still have outages and latency spikes.

Is there anything you can do on the client end if bufferbloat exists somewhere along the data path? I have the feeling that the Peruvian backend in general is overloaded and suffering from bufferbloat (latency spikes on mobile often go up to 3000-5000ms even with only a single video call on it even when speedtests are showing 20/5mbps on the same connection).

hkolvenbach, 2022-11-08:
By the way: thanks a lot for building this software, I tried to configure something similar by hand with OpenWRT before discovering your project and I didn't get far, so happy to sponsor your project through my company.

I wonder if the following makes it easier: I have a Starlink with much lower latency (30-300 ms), but due to my traveling I cannot always guarantee 0 obstructions, so have regular dropouts. Is there any combination of MPTCP algorithms & scheduling that could do a low-latency switchover to a 4G / Wifi when Starlink drops out?

Is it possible to set a per-link timeout for MPTCP, e.g. max 300 ms on Starlink until it is deemed offline, but >1s on the high latency links? I am most concerned about latency for video calls and do not care too much about throughput or data consumption. Would the `redundant` scheduler in MPTCP be the right choice in this case? Is it essentially duplicating packets over all links?

By the way, this is what my latency via MPTCP over the Peruvian 3 aggregated LTE lines looks like (tested with Google Meet). I have never seen such terrible network conditions before...
[image]
[image]
[image]

Ysurac, 2022-11-08:
You can increase Multipath TCP SYN retries in Network->MPTCP and try others Multipath TCP scheduler and TCP Congestion Control.
You can also try others VPN in System->OpenMPTCProuter, Wizard.

In all case, OpenMPTCProuter often doesn't perform well with high latencies and packet loss.

hkolvenbach, 2022-11-08:
Thanks for the quick reply! I will take a look at those settings and see if it helps

Network-Traditions, 2022-11-08:
Refer to #2594.  Also, the "Redundant" Multipath TCP scheduler in the Network-Settings tab goes a long way towards improving the quality of your internet connections at some expense in speed and increased bandwidth consumption.  We are still working with our 5G cellular provider (T-Mobile USA) trying to get our plan setup correctly to support 3rd Party devices.  We're deploying our setups for business service so quality and control of the connection is crucial so using our own hardware and software stack is required.

hkolvenbach, 2022-11-09:
@Network-Traditions thanks for the link and the detailed writedown! Your setup seems very similar indeed (minus the extreme latency variation down here) so I will try to adapt mine to your findings.

hkolvenbach, 2022-11-10:
@Network-Traditions I will try :-) It's a lot hit and miss and very unpredictable, but if I think I can identify trends I will definitely share them. What I did already notice is that OpenMPTCPRouter + Starlink is not working too well with obstructions > 25%, because the timeframe between obstructions is too short for OpenMPTCPRouter to pick up a live connection on Starlink (e.g. the 1s delay). With obstructions <25%, there are enough longer windows for OpenMPTCPRouter to detect an active connection and route accordingly. I have tried `redundant`, `BLEST`and `ECF` as schedulers, but can't really give any indication at the moment.

I did reverse
```
System-OpenMPTCProuter-Advanced Settings-IPv4 TCP SYN retries = 2
System-OpenMPTCProuter-Advanced Settings-IPv4 TCP SYN retries1 = 1
System-OpenMPTCProuter-Advanced Settings-IPv4 TCP SYN retries2 = 4
```
back to the defaults because with the high latency connections, it resulted in too many aborted connections and websites not loading even though a (very slow / high latency) connection was available.

hkolvenbach, 2022-11-10:
This is what the bandwidth distribution looks like, latency stays acceptable (but does spike above >1s) with a lot of ongoing downloads in the background. When Starlink does kicks in, it does pull some weight (`wgwwan`, `mikrotik-wlan`, and `2g-wifi` are all LTE connections with different providers, but widely fluctuating). Is there a tool to monitor latency and bandwidth in one graph to see if spikes are correlated?
[image]

Network-Traditions, 2022-11-10:
> Is there a tool to monitor latency and bandwidth in one graph to see if spikes are correlated?

Not that I'm aware of.  For close to real-time latency vs traffic monitoring, I've been using pfSense which sits between OMR and my intranet infrastructure.  pfSense "pinger" is set to ping the VPS via the OMR connection so I get 1 second resolution there while pfSense traffic graphs chart upload and download bandwidth:

[image]
Here you can see OMR to VPS RTT @ 105ms with a standard deviation of 32.4ms and 0% packet loss.  Since the Starlink and T-Mobile NOCs typically dump my connections in the southwest US, I encounter about 100ms of total RTT latency connecting with my VPS in central US, therefore, a 105ms ping is very good.  Sometimes I'll see it drop to about 75 when it's really good.  Under load I typcially see latency climb to 200-250ms.  I've adjusted my SQM to keep it to a max of 300ms when possible.

When fine tuning my connection for quality and speed, I watch this pfSense display and the OMR "Status" display together so I can get a sense of OMR WAN latency while observing the pfSense results.  While adjusting the configuration to maximize maximum throughput for synthetic benchmarks (speedtest.net, fast.com, omr speed testing, etc), I find real load performance typically does not perform as well.  If you review other issues I've been evaluating you'll notice I've been battling my T-Mobile connection.  Fingers crossed, I think I've finally made significant progress on that front as I believe the T-Mobile system was programmitically blocking and dropping my connections becasuse my equipment was not a T-Mobile supplied router, which is the rate plan I was originally setup with.  I believe in the last 24 hours, they've finnally got me on a plan that supports 3rd party devices and have not had my modem connection disconnect for the last day as a result.  It took awhile to flush that out and up to this point, I was constantly pursuing adjustments to OMR thinking that was my only option to improve my experience.

Lastly, I'll leave you with a recommendation to review #2588.  This could significantly increase the quality of your overall connection.  Based on the experience documented, I switched my scheduler to "redundant" and observed an increase in quality, reliability and higher "real load" throughput in my system.  Since I believe my T-Mobile connection should perform more reliably now, I'm switching back to "ECF" to reduce my montly usage of Starlink since they just introduced a 1TB traffic allotment afterwhich, my connection may be significantly throttled.  My hope is ECF will spread the overall traffic, ultimately reducing my Starlink usage comfortably below the new limit.  To date, I'm already noticing my latency sensitive traffic, specifically VOIP phone calls is suffering from jitter so it will be back to testing to attempt more traffic shaping to resolve this issue.  Good luck!

hkolvenbach, 2022-11-25:
> Here you can see OMR to VPS RTT @ 105ms with a standard deviation of 32.4ms and 0% packet loss. Since the Starlink and T-Mobile NOCs typically dump my connections in the southwest US, I encounter [...]
latency climb to 200-250ms. I've adjusted my SQM to keep it to a max of 300ms when possible.

Nice, these are really good numbers and should be fine for video calls. I am not sure which setting i changed, but I now get a lot of connection refused / connection dropped errors that I didn't have before... Maybe I need to go back to the defaults and start from scratch, next time I will document better what I changed ;-)

> Lastly, I'll leave you with a recommendation to review #2588. This could significantly increase the quality of your overall connection. Based on the experience documented, I switched my scheduler to [...]

Yeah, the new 1TB soft limit adds another uncertainty to the mix, so far I was lucky enough to not get close to it.

Network-Traditions, 2022-11-25:
The process of testing various configurations is challenging.  Since there are so many moving variables, many of which can't be properly quantified because of lack of access to necessary data, the evaluation of performance starts to have an anecdotal origin.  While I found synthetic benchmarking helpful to establish the anticipated impact of changing a single setting, the actual conclusion of said change is based on real world evaluation (i.e. video, voip, and other latency or bandwidth sensitive usage).  While it may be helpful to establish general guidelines for configuration to aggressively modify a number of settings per test, in the end, a singular, exhaustively real world test is what truly confirms the best practice configuration for each specific deployment.

As far as ISP consumption limits, our small business application consumes about 2+ TB per month.  When T-Mobile is working effectively and a load sharing scheduler is used, we will succeed in keeping our StarLink consumption below their 1TB limit.  Next summer when we expect to go mobile with our platform we anticipate T-Mobile will carry an increasing share of our total consumption using a 5G "Ultra-Capacity" connection vs our current typical straight 5G connection due to improving our distance and signal quality in relation to the connected cell tower.  We will keep posting our experiences to help the community and the project.

closed 2023-02-28 by github-actions[bot]