Ysurac/openmptcprouter #2653: Qualcomm MSM8916 LTE Dongle not supported

karma-rohit, 2022-11-09:
I bought this Qualcomm MSM8916 Chipset LTE Dongle, but it is not working on the OpenMPTCP Router router.
I also have a ZTE dongle which got easily detected on the router but this Qualcomm is not getting detected by the router.
I noticed that when I connect this dongle to Router, it shows a MAC address for 2 seconds in the DHCP Interface and then disconnected.
The ZTE dongle is also an RNDIS device and got detected as an 'eth1' device when connected to the USB port, but the Qualcomm MSM8916 Dongle does not show any port occupied when connected.
Note: The dongle is working fine on Windows PC as RNDIS.

## Specifications

  - OpenMPTCProuter version: v0.59.1
  - OpenMPTCProuter VPS version: Latest (updated recently)
  - OpenMPTCProuter VPS provider: Google Cloud
  - OpenMPTCProuter platform: Raspberry Pi 4B 64bit

Network-Traditions, 2022-11-09:
Are you able to change the mode of the dongle?  For our setup using a [Telit FN980 5G Modem (HWv1.0 )](https://www.telit.com/devices/fn980-and-fn980m-data-cards-support-5g/) in a [USB3.0 M.2 Key B Modem Adapter Enclosure](https://thewirelesshaven.com/shop/mini-pcie-m2-adapters/modem-enclosure/usb3-0-to-ngff-m-2-key-b-4g-5g-modem-adapter-enclosure-with-sim-card-slot-new-style/) it supported QMI, MBIM, NCM and RNDIS modes of operation.  We tried all four and ended up getting the best results using ModemManager with the modem in QMI mode (see #2359 - reliability has improved significantly after upgrading to OpenMPTCProuter: v0.59.1-5.4 see #2594), which is recommended by Telit.

karma-rohit, 2022-11-10:
The Dongle I'm using is a cheap one, having a Qualcomm chipset and no inbuild mode change option through their web interface. Maybe, it can possible through AT Commands but the AT Command Tester is also not detecting the dongle.
I tested 'Static Address' mode on OMR and got some clue. The device got detected for few seconds "without any uplink/downlink" and then through error as 'Unknown error (DEVICE_CLAIM_FAILED).
I found these threads on this issue:
https://github.com/openwrt/openwrt/issues/7795
https://github.com/Ysurac/openmptcprouter/issues/1948

I'm testing with two modems. One is ZTE chipset which works fine on DHCP mode and got connected without issue. But the Qualcomm is making problem. I think it's due to time it take to switch between storage to cellular (not sure).
In thread link above, other user facing the same problem and with same Qualcomm chipset.
[image]

Network-Traditions, 2022-11-10:
With my setup, I use the "minicom" package installed within the System-Software tab of OMR so I can use AT commands to configure the modem from the console (/dev/ttyUSB3 in my case). Check out this [YouTube video](https://youtu.be/DRddwfZ_TBY?t=239) for a demonstration.

karma-rohit, 2022-11-11:
So, I'm not able to make the connection to the dongle. Below is the error log from different queries.

`root@OpenMPTCProuter:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=03 Prnt=03 Port=01 Cnt=01 Dev#=  5 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=05c6 ProdID=9024 Rev=ff.ff
S:  Manufacturer=Android
S:  Product=Android
S:  SerialNumber=123456789321
C:* #Ifs= 2 Cfg#= 1 Atr=80 MxPwr=500mA
A:  FirstIf#= 0 IfCount= 2 Cls=e0(wlcon) Sub=01 Prot=03
I:* If#= 0 Alt= 0 #EPs= 1 Cls=e0(wlcon) Sub=01 Prot=03 Driver=(none)
E:  Ad=82(I) Atr=03(Int.) MxPS=   8 Ivl=32ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=(none)
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms`

`root@OpenMPTCProuter:~# dmesg | grep rndis_host

[   11.850666] rndis_host 1-1.2.2:1.0: no of_node; not parsing pinctrl DT
[   11.852308] rndis_host 1-1.2.2:1.0 usb0: register 'rndis_host' at usb-0000:01:00.0-1.2.2, RNDIS device, 26:3e:e0:c6:ae:1c
[   11.863402] usbcore: registered new interface driver rndis_host
[   35.655826] rndis_host 1-1.2.2:1.0 usb0: unregister 'rndis_host' usb-0000:01:00.0-1.2.2, RNDIS device
[   36.145161] rndis_host 1-1.2.2:1.0: no of_node; not parsing pinctrl DT
[   36.146769] rndis_host 1-1.2.2:1.0 usb0: register 'rndis_host' at usb-0000:01:00.0-1.2.2, RNDIS device, c2:60:63:2b:4b:09
[   36.159496] rndis_host 1-1.2.2:1.0 usb0: unregister 'rndis_host' usb-0000:01:00.0-1.2.2, RNDIS device`

`Status > System Log

Fri Nov 11 12:12:06 2022 kern.info kernel: [  415.359314] usb 1-1.4: new high-speed USB device number 4 using xhci_hcd
Fri Nov 11 12:12:06 2022 kern.info kernel: [  415.464218] usb 1-1.4: New USB device found, idVendor=05c6, idProduct=9024, bcdDevice=ff.ff
Fri Nov 11 12:12:06 2022 kern.info kernel: [  415.472575] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Fri Nov 11 12:12:06 2022 kern.info kernel: [  415.479885] usb 1-1.4: Product: Android
Fri Nov 11 12:12:06 2022 kern.info kernel: [  415.483718] usb 1-1.4: Manufacturer: Android
Fri Nov 11 12:12:06 2022 kern.info kernel: [  415.487985] usb 1-1.4: SerialNumber: 123456789321
Fri Nov 11 12:12:06 2022 kern.debug kernel: [  415.492857] usb 1-1.4: no of_node; not parsing pinctrl DT
Fri Nov 11 12:12:06 2022 daemon.notice netifd: Interface 'Dongle1' is enabled
Fri Nov 11 12:12:06 2022 daemon.notice netifd: Network device 'usb0' link is up
Fri Nov 11 12:12:06 2022 daemon.notice netifd: Interface 'Dongle1' has link connectivity
Fri Nov 11 12:12:06 2022 daemon.notice netifd: Interface 'Dongle1' is setting up now
Fri Nov 11 12:12:06 2022 kern.debug kernel: [  415.495347] rndis_host 1-1.4:1.0: no of_node; not parsing pinctrl DT
Fri Nov 11 12:12:06 2022 kern.info kernel: [  415.496972] rndis_host 1-1.4:1.0 usb0: register 'rndis_host' at usb-0000:01:00.0-1.4, RNDIS device, 86:f6:28:d2:82:a4
Fri Nov 11 12:12:06 2022 kern.info kernel: [  415.509375] rndis_host 1-1.4:1.0 usb0: unregister 'rndis_host' usb-0000:01:00.0-1.4, RNDIS device
Fri Nov 11 12:12:06 2022 daemon.notice netifd: Network device 'usb0' link is down
Fri Nov 11 12:12:06 2022 daemon.notice netifd: Interface 'Dongle1' has link connectivity loss
Fri Nov 11 12:12:06 2022 daemon.notice netifd: Interface 'Dongle1' is now down
Fri Nov 11 12:12:07 2022 daemon.notice netifd: Interface 'Dongle1' is disabled
Fri Nov 11 12:12:07 2022 daemon.warn [4502]: <warn>  [usb0] invalid sysfs path read for net/usb0
Fri Nov 11 12:12:07 2022 daemon.warn [4502]: <warn>  [base-manager] couldn't handle kernel event: device net/usb0 not found`

Don't know what's going on.

Network-Traditions, 2022-11-11:
Making a serial connection in order to send AT commands is required to leverage the full functionality of a modem.  If the device does not support such a connection, then it's likely that capability does not exist or is not exposed.  OEM documentation and/or technical support would be the best place to start if it exists.  While we were able to get our modem connected to OMR using RNDIS, the performance was extremely slow and unusable.

karma-rohit, 2022-11-18:
After digging through the internet for days, I found the solution.
Here https://forum.openwrt.org/t/3g-4g-usb-dongle-not-showing-up/33926/16

Thanks to - https://forum.openwrt.org/u/bmork and https://forum.openwrt.org/u/fiatjaf

closed 2022-11-18 by karma-rohit

iamdivyanshverma, 2023-06-07:
Hi @karma-rohit  I am facing the same issue Can you please help me how you solved your issue, i have checked the above link but coudnot understand much , I am new to openwrt .

Thanks in Advance

karma-rohit, 2023-06-08:
> Hi @karma-rohit I am facing the same issue Can you please help me how you solved your issue, i have checked the above link but coudnot understand much , I am new to openwrt .
> 
> Thanks in Advance

If the issue is ditto as what I had encountered, then the solution is to first look for your device in the mode.json file of the openwrite router. 
If it was not in the list (which has greater chance) than add it in the list and set the mode switch time so that it got switched accordingly from storage to modem before getting disconnected by time-out.

All this process is explained in the given link above and you need to follow them from the beginning, there's no bypass of that. If you are here then I'm sure you know little bit of coding.

bilogic, 2024-12-23:
@karma-rohit did you manage to remove the NAT on the dongle and go into QMI or MBIM?