Ysurac/openmptcprouter #2654: OMR compilation error on x86_64 v0.59rc3

satmarin, 2022-11-09:
## Expected Behavior
Hi I have been trying to compile the software for use on x86_64 platform but keep getting an error with golang

## Current Behavior
On Debian 10
Have used build.sh from /openmptcprouter folder and also make from .../5.4/source but error remains (see below)

...
net/http/pprof
net/http/fcgi
net/rpc
net/rpc/jsonrpc

( cd "/home/qsecofr/openmptcprouter/x86_64/5.4/source/build_dir/hostpkg/go-1.17.3/src" ; GOROOT_FINAL="/home/qsecofr/openmptcprouter/x86_64/5.4/source/staging_dir/hostpkg/lib/go-cross"  GOROOT_BOOTSTRAP="/home/qsecofr/openmptcprouter/x86_64/5.4/source/build_dir/hostpkg/go-1.17.3/.go_bootstrap" GO_LDFLAGS="-buildmode pie" GOCACHE="/home/qsecofr/openmptcprouter/x86_64/5.4/source/tmp/go-build" GOENV=off CC="gcc" CXX="g++"  bash make.bash --no-banner ; )
Building Go cmd/dist using /home/qsecofr/openmptcprouter/x86_64/5.4/source/build_dir/hostpkg/go-1.17.3/.go_bootstrap. (go1.4-bootstrap-20170531 linux/amd64)
Building Go toolchain1 using /home/qsecofr/openmptcprouter/x86_64/5.4/source/build_dir/hostpkg/go-1.17.3/.go_bootstrap.
Building Go bootstrap cmd/go (go_bootstrap) using Go toolchain1.
Building Go toolchain2 using go_bootstrap and Go toolchain1.
Building Go toolchain3 using go_bootstrap and Go toolchain2.
go build cmd/compile/internal/ssa: /home/qsecofr/openmptcprouter/x86_64/5.4/source/build_dir/hostpkg/go-1.17.3/pkg/tool/linux_amd64/compile: signal: killed
go tool dist: FAILED: /home/qsecofr/openmptcprouter/x86_64/5.4/source/build_dir/hostpkg/go-1.17.3/pkg/tool/linux_amd64/go_bootstrap install -gcflags=all= -ldflags=all=-buildmode pie -a -i cmd/asm cmd/cgo cmd/compile cmd/link: exit status 1
make[3]: *** [Makefile:370: /home/qsecofr/openmptcprouter/x86_64/5.4/source/build_dir/hostpkg/go-1.17.3/.built] Error 2
make[3]: Leaving directory '/home/qsecofr/openmptcprouter/feeds/openmptcprouter/golang/golang'
time: package/feeds/openmptcprouter/golang/host-compile#251.85#59.54#538.42
    ERROR: package/feeds/openmptcprouter/golang [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/openmptcprouter/golang/host/compile] Error 1
make[2]: Leaving directory '/home/qsecofr/openmptcprouter/x86_64/5.4/source'
make[1]: *** [package/Makefile:110: /home/qsecofr/openmptcprouter/x86_64/5.4/source/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/qsecofr/openmptcprouter/x86_64/5.4/source'
make: *** [/home/qsecofr/openmptcprouter/x86_64/5.4/source/include/toplevel.mk:230: world] Error 2
qsecofr@deb10dev:~/openmptcprouter/x86_64/5.4/source$

Thank you for sharing your thoughts on this and how to fix it

Michaël

## Specifications

  - OpenMPTCProuter version: v0.59rc3
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter VPS provider: N/A
  - OpenMPTCProuter platform: x86_64

Ysurac, 2022-11-10:
"compile: signal: killed" look on your computer what it's killed (memory, cpu,...)

SatExo, 2022-11-10:
Thank you very much,

My Debian VM was apparently running out of memory, After closer inspection it seems it had approx 650MB RAM detected where the available was 12GB.

Turns out the dynamic memory allocation on Hyper-V is blowing up without actually becoming available on the Debian VM.  
Set it to fixed memory in Hyper-V and worked like a charm.

Hope this helps someone some day

M.

closed 2022-11-13 by Ysurac