Ysurac/openmptcprouter #2656: Crash Wan couldn't handle kernel event: device net/SQM_IFB_7e1f5 not found

hle5128, 2022-11-10:
## Expected Behavior
all wan up and running
all wan stay uplink instead

## Current Behavior
all wan uplink then crash downlink every second. 
fresh installed and wan up and down every second
yes, monitor live log, found
daemon.err modprobe: failed to find a module named sch_fq_codel 

## Possible Solution
kernel update?

## Steps to Reproduce the Problem

  1. fresh install openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined-efi.img.gz on   Intel N3700, 4NIC, WAN1/eth1 and WAN2/eth2 are connected to individual 5G
  2. after setup DHCP in each WAN, each wan starting to crash every second

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4-r0+16594
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1028
  - OpenMPTCProuter VPS provider: BUYVM
  - OpenMPTCProuter platform: x86_64 Intel N3700 4x 2.5GbE NIC Ports
  - Country: USA 

https://pastebin.com/Wvztcb6y

Ysurac, 2022-11-10:
If you use Intel I225 2.5GbE look at https://github.com/Ysurac/openmptcprouter/issues/2584

Ysurac, 2022-12-22:
It's related to SQM virtual interfaces, not a problem.

closed 2023-03-27 by github-actions[bot]