Ysurac/openmptcprouter #2657: cznic_turris-omnia-initramfs-kernel.bin

MTSmhas, 2022-11-10:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

**Describe why you are asking for this support?**

**Additional context**

Hi can you Build a Image for this device -> https://openwrt.org/toh/turris/turris_omnia
or a description of how to install it on openwrt. 

Thanks

Ysurac, 2022-11-24:
You can build it yourself: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform
(I only support officially platform that I own)

MTSmhas, 2022-11-25:
This dont work but we have build a own image with openwrt and your feeds and this works, but we have a bug on all devices... after we start a speedtest on speedtest.net is the Admin Script not available for 2 seconds and the omr track restart the connections.

Ysurac, 2022-11-25:
OpenMPTCProuter is not only the feed, there is MPTCP patch and configuration in the build repo, without this you can't get any aggregation.

MTSmhas, 2022-11-25:
yes we have hack the kernel... now the error is on your last build available. And it is only if the upload test run.
[image]

MTSmhas, 2022-11-25:
we have the error on your original x86 original build. and on own custom build.

Ysurac, 2022-11-25:
You need to enable SQM and set download/upload speed.

MTSmhas, 2022-11-25:
Ok on which device? SQM is enable.

Ysurac, 2022-11-25:
On all devices, if there is no changes, set a lower upload speed.

MTSmhas, 2022-11-25:
Okay we test it. Thanks ;)

MTSmhas, 2022-11-25:
Perfect thanks it work. oh man so simple xD

closed 2022-11-25 by MTSmhas