Ysurac/openmptcprouter #2658: How to speed up compilation after the first compilation

atyouspace, 2022-11-11:
## Expected Behavior
After the first time `make`, the following make should be faster.

## Current Behavior
When I compile by `make` for the first time, it spend me more than 24 hours. However, if I compile again, it still spends me very long time.

what should I do? Normally, the following make should be faster.

## Specifications

  - OpenMPTCProuter version: 0.59.1
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: arm

Kalimeiro, 2022-11-11:
it's indicate in the doc, you need to use the J option and specifiy the number of cpu thread you want to compile faster.

atyouspace, 2022-11-11:
> 

I mean if I need to MAKE again.
Normally, only the first MAKE takes a particularly long time, because what has already been compiled should be ignored.
However, I found that every time I MAKE, it feels like I'm starting all over again.

@Ysurac

Ysurac, 2022-11-11:
All is not compiled again if you use make in source subdirectory.
If ti make more than 24hours than use another computer, you have a really slow computer... Even Github Actions doesn't take so much times.

atyouspace, 2022-11-14:
Yes, use -j parameter will speed up.

closed 2022-11-14 by atyouspace