Ysurac/openmptcprouter #2661: OMR-Bypass don't work

n4sty3us, 2022-11-11:
## Expected Behavior
Hello, I entered the MAC-Address of my server (connected via ethernet cable on the LAN interface) on the OMR-Bypass page to go out on the internet using the modem connection connected to the WAN interface

## Current Behavior
By enabling the bypass, the server does not go to the internet. The problem occurs both if you select a specific interface and if you leave it at none.

My configuration:
[image]
[image]

Could I have information on how to solve the problem? Thank you!!!

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4
  - OpenMPTCProuter platform: Linksys WRT32X

Kalimeiro, 2022-11-12:
i can't reproduced this issue on 0.59.2 alpha, i have bypass my computer mac to a random wan and internet go correctly through this wan.

n4sty3us, 2022-11-12:
I don't think wan's name is meaningful about the problem.  I have already tried not to select any interface (as indicated, assign the master one) ... I try to assign them wlan1 but my interest is to go out on the wan interface.

How do I restart omr-bypass?

n4sty3us, 2022-11-14:
Update: I have changing the interface (wlan1) but the pc still doesn't go internet

kevinh-csalabs, 2022-11-16:
Are you able to ping the wan? if not where does traceroute fail?

n4sty3us, 2022-11-16:
If the bypass is not enabled, the devices connected to the lan/wifi work correctly and use both connections on the wan port (the speedtest returns the sum of the two connections).

Where to ping the wan port from? Should it be done with the bypass rule active?

kevinh-csalabs, 2022-11-16:
from the device you want to go directly via that wan. The bypass should be enabled.
If you can't ping the wan from the device try from the router via ssh. If neither can ping the wan then there is a problem with the wan. If only the device trying to bypass can't ping then something is getting blocked and traceroute should help determine where that is happening.

n4sty3us, 2022-11-16:
Ok, so I ping both from the pc with the bypass active and directly from the Linksys. Do I have to point the IP that is indicated in the wan interfaces page 192.168.178.31?

kevinh-csalabs, 2022-11-16:
are you using a linksys for openmptcprouter? If so, then yes. If no then it sounds like maybe you have multiple routers and this is the issue? 

Yes you will use the wan from the interfaces page.

n4sty3us, 2022-11-16:
My system has a Linksys WRT32X with installed OpenMPTCProuter. FRITZ!Box 7530 connected via ethernet to the wan port and ZTE MF289F connected in wifi to the wlan1 port

n4sty3us, 2022-11-18:
I did a reset and even assigning static IPs to the two WAN ports, I still have the same problem. From the PC with active bypass, do not ping the private or public IP of the WAN port

datapharmer, 2022-11-18:
1. what does traceroute show from the client that can't ping the lan? This should show where the connection is getting stuck
2. if you use bypass for a specific website that accepts ping (like google) does bypass work properly for that client (you can confirm ip seen on google or use traceroute to confirm it is using the wan and not the vps)? 

It looks like your wan addresses are local which means you have at least a double nat which isn't great. If you can use bridge mode for the wan connections that may solve some issues.

n4sty3us, 2022-11-18:
@datapharmer I put the google service in the bypass:
[image]

This is the result of the ping/traceroute:
`C:\Users\COLACEC2>tracert google.com

Traccia instradamento verso google.com [142.250.180.174]
su un massimo di 30 punti di passaggio:

  1     *        *        *     Richiesta scaduta.
  2     *        *        *     Richiesta scaduta.
  3     *        *        *     Richiesta scaduta.
  4     *        *        *     Richiesta scaduta.
  5     *        *        *     Richiesta scaduta.
  6     *        *        *     Richiesta scaduta.
  7     *        *        *     Richiesta scaduta.
  8     *        *        *     Richiesta scaduta.
  9     *        *        *     Richiesta scaduta.
 10     *        *        *     Richiesta scaduta.
 11     *        *        *     Richiesta scaduta.
 12     *        *        *     Richiesta scaduta.
 13     *        *        *     Richiesta scaduta.
 14     *        *        *     Richiesta scaduta.
 15     *        *        *     Richiesta scaduta.
 16     *        *        *     Richiesta scaduta.
 17     *        *        *     Richiesta scaduta.
 18     *        *        *     Richiesta scaduta.
 19     *        *        *     Richiesta scaduta.
 20     *        *        *     Richiesta scaduta.
 21     *        *        *     Richiesta scaduta.
 22     *        *        *     Richiesta scaduta.
 23     *        *        *     Richiesta scaduta.
 24     *        *        *     Richiesta scaduta.
 25     *        *        *     Richiesta scaduta.
 26     *        *        *     Richiesta scaduta.
 27     *        *        *     Richiesta scaduta.
 28     *        *        *     Richiesta scaduta.
 29     *        *        *     Richiesta scaduta.
 30     *        *        *     Richiesta scaduta.

Traccia completata.

C:\Users\COLACEC2>ping 142.250.180.174

Esecuzione di Ping 142.250.180.174 con 32 byte di dati:
Richiesta scaduta.
Richiesta scaduta.
Richiesta scaduta.
Richiesta scaduta.

Statistiche Ping per 142.250.180.174:
    Pacchetti: Trasmessi = 4, Ricevuti = 0,
    Persi = 4 (100% persi),

C:\Users\COLACEC2>`

I have seen that ping or traceroute doesn't work from clients even without bypass.

I confirm that the two modems are not in bridge mode

kevinh-csalabs, 2022-11-18:
If ping and traceroute to something like google doesn't work for clients not in bypass then you have some routing problem unrelated to omr-bypass. Check your gateway is correct and client has a valid ip for the router as these don't seem to be even making it to the router or the vps.

n4sty3us, 2022-11-18:
The clients connected to the router work perfectly on the internet and the traffic is correctly routed on the VPS… the dhcp server has been activated on the lan and the clients get the ip dynamically from the gateway

kevinh-csalabs, 2022-11-18:
Not sure how to help you if traceroute isn't working as this feature works as I'd expect for me.

Traceroute without OMR-BYPASS should show:
1. OpenMPTCProuter
2. VPN IP (typically 10.255.254.1)
3. VPS IP

Traceroute with OMR-BYPASS should show:
1. OpenMPTCProuter
2. WAN IP (master interface or selected interface)

If you aren't getting either of those then there is something wrong with your general network configuration the router is advertising or with the network configuration getting set on the client. 

I'd suggest comparing ipconfig /all and route print with and without the bypass enabled and compare.

n4sty3us, 2022-11-18:
as suggested by another user, can the presence of non-bridged modems be the cause of this anomalous behavior?  if I connect the vdsl modem in bridge, the wan port of the router must always be static right?

Kalimeiro, 2022-11-19:
hi, i have ask you why your wan connection has no number like wan1 wan2 wan3 when you create a rules in omr-bypass, like this : 

[image]

I think it's not possible to bypass to an interface not correctly named, so, i suggest to test with only one WAN connected to an ethernet port (not your USB LTE MODEM), you need to get WAN1 when you create an omr-bypass rules, then, just try if bypass works. Use the "Settings wizard" to setup your omr.

n4sty3us, 2022-11-19:
@Kalimeiro I used the wizard and got this:
[image]

by default, OMR sets the wan port as LAN and the lan ports (1,2,3,4) as WAN.  I changed this configuration because I don't need to have 4 ethernet WAN ports while I need to use 4 LAN ports for my network.  for this reason I created a bridge called "br-lan" where I inserted the 4 lan ports.

the really strange thing is that all the clients (pc, smart tv, smarthphone, iot, etc.) connected to the OMR router are working correctly on the internet (they all have the IP of the VPS and the total speed is the aggregation of the two internet lines ).  bypass, ping and traceroute don't work at all

@Ysurac could it be an OMR bug on WRT32X?

kevinh-csalabs, 2022-11-19:
You’ve got the wan1 and lan on the same port set this is probably messing up your firewall and blocking the route from lan to wan directly. 

Sort out your lan and wan1 issues. This may be a limitation on your device as some switch ports on routers are “soft” and really more of a hub from a hardware perspective which might be the problem.

n4sty3us, 2022-11-19:
@kevinh-csalabs how can I make this fix? I have no skill on OMR