Ysurac/openmptcprouter #2664: Can't get Firstnet connection to work (RPI4)

adamb, 2022-11-13:
## Expected Behavior

I have an Firstnet LTE connection using a Netgear nighthawk.  I've turned off the wifi and set the IP to 10.1.1.1.  I can't seem to make the Netgear use 192.168.10.1 as recommended in the documentation.

## Current Behavior

I have changed the settings in wizard to 10.1.1.1 and 10.1.1.2.   I've turned off the DHCP on the Netgear.  I'm seeing *No IP Detected" for the wan1 connection.  Wan2 is starlink but I don't have it plugged into the router yet.  I'm going to add that once I get the wan1 working.

I have the Pi, the Netgear LTE modem, and my mac all plugged into a tp-link TL-SG108 desktop switch.  I have noticed that my mac can't get an DHCP IP lease from  the PI if the netgear is plugged into the switch.  When I unplug it, the mac will get an IP.  If I manually set my IP to 10.1.1.42 I can then connect directly to the Netgear at 10.1.1.1 and configure it (except it won't let me change it's IP to 192.168.10.1 for some reason...)

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4 
  - OpenMPTCProuter VPS version: 0.1022.5.4.74-mptcp
  - OpenMPTCProuter VPS provider: self (it's at my house where there is fiber internet)
  - OpenMPTCProuter platform: RPI4

[image]

Ysurac, 2022-11-13:
You seems to have set a gateway on interface but no IP. IP doesn't seems to be defined for wan1 and wan2, check the wizard.

adamb, 2022-11-13:
The problem was I was using wan1 when I should have been using eth0 with a MacVLAN.

Now that I have that working, I need to figure out how to add wan2 connected to Starlink in bridge mode.

closed 2022-11-13 by adamb