Ysurac/openmptcprouter #2665: Can OMR use Starlink in Bridge Mode?

adamb, 2022-11-13:
## Expected Behavior

I'm using an LTE router and would like to add a connection over Starlink.  Starlink is in bridge mode though. So not sure how to put that into the wizard.  I tried DHCP and Bridge but those didn't seem to work.

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 1028 5.4.207
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: RPI4

adamb, 2022-11-13:
I found this in another issue:
https://github.com/Ysurac/openmptcprouter/discussions/1756#discussioncomment-2838697

Says to use DHCP.  But I have just one switch and the Pi is a DHCP server as well as the Starlink in bypass mode (CGNAT?) so having two DHCP servers on the same network seems bad.  

Maybe it's not possible using a Pi with just one ethernet port.

nbald, 2022-11-13:
I tried and could not make it work.
I ended up placing a small router between the dish and my OMR network - got rid of the starlink wifi router, of course. Starlink -> DHCP -> router NAT -> OMR

Network-Traditions, 2022-11-14:
We've been using StarLink V2 in "bypassed" bridged mode for several months now.  2 DHCP servers on the same **_subnet_** is undesirable so make sure Starlink is the only DHCP server connected to a subnet where your OMR router is the only WAN DHCP client as the Starlink router will only have one DHCP address available (only one device will succeed in obtaining the DHCP address).  Take a look at #2584 with a specific focus on the "Force link" setting (Network->Interface->Edit->Advanced Settings) and also take a look at https://nelsonslog.wordpress.com/2021/04/07/openwrt-vs-starlink-dhcp-leases.

datapharmer, 2022-11-16:
Yes I use starlink v2 in bypass without issue. I suggest investing in a switch with some decent functionality and you can do something to prevent the dhcp handout from getting anywhere but the pi such as enabling dhcp snooping, port isolation rules or vlans. The goal is to keep switch ports other than the pi from talking to your other wan connections.

nbald, 2022-11-16:
I use a MikroTik Hex Lite switch. 50$ . Works perfectly

adamb, 2022-11-16:
I have an Edge Router X. I wonder if that would work. Will investigate.  

Also, I'm worried about the whole system going down if either my VPS or Pi fail.  Maybe I can use the ERX to fail over if the Pi connection goes south.  I could fail back to load balancing over the Starlink and my LTE connection if the Pi stops working.

datapharmer, 2022-11-17:
not sure how stable the router firmware is on it, but in theory it should as you can just use the different ports for different wans to provide isolation.

Multiple vps is already supported for failover, and opwnwrt has instructions for high availability if you want local router redundancy: https://openwrt.org/docs/guide-user/network/high-availability

Keep in mind that there's always a tradeoff between redundancy and complexity.

adamb, 2022-11-23:
I have this working now.  Thanks for all the help.  I bought two E300 USB/ETH adapters so the Starlink can do DHCP and have it's own interface on the RPI4.  I'm using the built in ETH0 connection attached to my home Eero.  I use the two USB3 connections, one for the LTE (firstnet) and one for the Starlink.  

I do have two different VPS servers I'm using. One is at Linode and the other is at another residence that has a gigabit fiber internet service.  I'm experimenting to see which of these are better.   The advantage of using a server at a residence is that it avoids all the bypass issues for streaming services since it's a residence.

closed 2022-11-23 by adamb