Ysurac/openmptcprouter #2669: RUTX12 LAN/WAN & DHCP setup issues

DylanPW, 2022-11-18:
## Expected Behavior

1. Have eth0 ports (ports marked with LAN on the back of the router) as LAN ports
2. Enable DHCP for applicable interface
3. Have both modems report successful connections.

## Current Behavior

1. I'm unable to assign LAN to eth0, I have tried removing all of the WAN interfaces but that doesn't appear to resolve the issue. After applying change I am no longer able to connect to router.
2. After install, eth1 DHCP does not appear to work correctly out of the box, I have confirmed the force link and DHCP server options but I'm unable to get an IP address.
3. After setup, one modem reports as having GATEWAY DOWN; I am unable to get both modems to connect to the VPS server.

I'm unsure of whether the DHCP/LAN issues are a config error on my end or if they are the result of a bug, if I need to do something to allow the use of the LAN ports and DHCP please advise what needs to be changed out of the box. I've had success running OMR off both a pi 4b and an x86 based firewall appliance, but am unable to get the config right on the RUTX12.

Thanks in advance

## Specifications

  - OpenMPTCProuter version: 0.1028
  - OpenMPTCProuter VPS version: 0.59.1-5.4-r0+-16594
  - OpenMPTCProuter VPS provider: Microsoft Azure
  - OpenMPTCProuter platform: Teltonika RUTX12

Network-Traditions, 2022-11-18:
[RUTX12](https://wiki.teltonika-networks.com/view/RUTX12) is an interesting device.  As such, it will present its unique challenges to successfully deploy it with the OpenMPTCProuter operating system.  @Ysurac posted the following in #2028:

> On the RUTX, the LAN port after installing OpenMPTCProuter is the port labelled as "WAN". This port have ip 192.168.100.1 and DHCP enabled by default.

That being said, he warned of RPi3 performance experience:

> This can work but still same CPU, a "Quad-core ARM Cortex A7, 717 MHz" with "256MB" RAM, so you can't expect a good speed with that. Maybe same result as a RPI3 with some luck if there is hardware AES.

Subsequently, if slow performance is not an issue for you, it seems some success has been achieved in deploying OpenMPTCProuter as documented in the aforementioned issue.  Good luck!

highfreq, 2022-11-19:
Had same problem when i tried to move lan to eth0, i locked myself out. What worked for me was remove all the wans and remove all the port configurations from the switch setup (tagged ecc ecc).
Let me know if it worked for you.
Didn't have to do anything to heva the modems connect at startup.
I have been using it for months now with no probs at all, plenty fast for my application.

DylanPW, 2022-11-21:
When you say you removed all port configurations from the switch setup, did you remove the WAN interfaces or have to tinker with the VLANs as well? Not super familiar with this aspect of openwrt/omr sorry, I appreciate the help very much though! 

Did you also have DHCP not working on the WAN port from a fresh install? I'm not sure if this is an issue on my end specifically as I can't see any reason why it wouldn't be working here.

Thanks!

highfreq, 2022-11-21:
I removed all the vlan configurations from switch.

I had some probs with DHCP on lan, i disabled it since i didn't need it and enabled it on wifi interface wich i added to same firewall zone with lan.

DylanPW, 2022-11-22:
Thanks for the pointer on the vlan configurations from the switch, that helped getting LAN on eth0.

For anyone having the same issue, I ended up having to manually edit `/etc/config/dhcp` as `option dhcpv4` was configured as disabled, rather than configured as `option dhcpv4 'server'`

Still troubleshooting poor throughput when bonding WAN connections as I'm getting worse speeds with two modems running compared to just one under stock fw.

highfreq, 2022-11-22:
Did you erase all vlan configuration and it worked for you too?

DylanPW, 2022-11-22:
Yeah, removing the vlan config completely solved moving LAN to eth0, but not having working DHCP - that needed the config tweak.

closed 2023-02-26 by github-actions[bot]