Ysurac/openmptcprouter #2671: Google DNS override

sp2003, 2022-11-20:
Hello
after the latest fw update of my Fire stick TV, it is no longer possible to change the first DNS which by default is 8.8.8.8
You can add 2 more personal dns, for 3 dns in total, but the first one is google's.
As a result omr-bypass doesn't work for disney+.

I've tried manually adding the protocols and the fire stick's IP and domain, but to no avail.

Is it possible to create a firewall rule that blocks google dns to a single device?

Ysurac, 2022-11-21:
You can block Google DNS, for example with a stick with IP 192.168.100.6, you need to add in Network->Firewall and specific rules tab:
```
iptables -I FORWARD -s 192.168.100.6 --destination 8.8.8.8 -j REJECT
iptables -I FORWARD -s 192.168.100.6 --destination 8.8.4.4 -j REJECT
```

sp2003, 2022-11-21:
with these rules, the fire stick does not detect internet.  is it possible to redirect to the gateway instead?

closed 2023-02-26 by github-actions[bot]