Ysurac/openmptcprouter #2676: Self-host VPN server using mini pc

ffrrzz, 2022-11-22:
Hi,

I have not been able to find a completely suitable VPN server for my location (Chile) at a less than 10 USD budget. The best one I tried only provides upload speeds of up to 250 mbps (usually in the 100 to 200 mbps range).

Can I put an atom mini-pc at a friend’s house? I can request a public ip address at that place, but the server would have to live behind NAT at the DMZ of the router. Would it be an issue to use a DDNS server (dynamic ip)?

If my friend’s connection ever goes down, what is the default behavior of the OMR router when it can’t connect with the VPS? Would it simply failover to a direct connection to one WAN as a failover?

 Bandwidth would not be a limit as I would be able to get symmetrical fiber internet at that place.

Thanks!

kevinh-csalabs, 2022-11-23:
If it is a true public ip and not CGNAT and there are no ISP port restrictions then yes, but that is a big IF. Pretty sure DDNS isn't supported at this time and it would mean your internet goes out briefly every time it changes. I would strongly suggest seeing if you can get a static IP.

Note most residential (even fiber) isn't guaranteed bandwidth - it is 'up to' and this would probably violate your friend's terms of service.

Default behavior is to fail over to master once vps/vpn down is detected.

ffrrzz, 2022-11-23:
Thanks for gesting back to me!

The ISP I would use is Entel. Their default config is CGNAT, but you are able to get a true dynamic public IP for free after getting in touch with their customer service.  Dynamic IP leases usually stay the same for weeks/months so IP changes would really be an issue.

I understand that all residential ISPs only advertise "up to" speeds/bandwidth, but local ISPs must publish the average speed they are able to provide: https://www.entel.cl/indicadores/red-fija/fibra-optica/ 
Fiber operators here tend to meet/exceed their advertised speeds. Consumer complains for those connections are mostly linked to poor wifi coverage (the go-to building material is reinforced concrete), or total backouts due to physical damage to the fiber link.

Breaking the TOS of the ISP would not be a problem. Local regulations say "[ISPs] may not arbitrarily block, interfere, discriminate, hinder or restrict the right of any Internet user to use, send, receive or offer any content, application or legal service through the Internet, as well as any other type of activity or legal use done through the network."

I guess I will give it a try. A small intel box with a AES-NI capable processor goes for USD 40 after discounts/coupons/taxes on Amazon. :-)

adamb, 2022-11-24:
I'm doing this and it's working well.  I bought a fitlet2 box and I have this stashed under the TV in my AirBNB property.  The internet is provided by Century Link and is about 1gb/s.  I put  the _vps_ in the DMZ on the router.  It's IP address does change occasionally, so I'll have to deal with that.  (This is also why I filed [this issue](https://github.com/Ysurac/openmptcprouter/issues/2680) since I'm curious how OMR handles VPS outages.)

I setup a something that updates the DNS entry so I can ssh back into it if I'm not around.  I also setup pingdom monitoring so I know when or if it goes down.  It's been pretty rock solid in terms of uptime for two years!  Phoenix has pretty good electrical service and Century Link's connection has also had impressive uptime.  (My main house is in Puerto Rico, so very different reliability stats!)

Hope this helps!

ffrrzz, 2022-11-24:
@
[image]

Thank @adamb for that data point! 
A pair of these boxes are on my way to test drive this approach. They appear to strike a good price/performance/power usage balance.

closed 2023-02-27 by github-actions[bot]