Ysurac/openmptcprouter #2678: IPV6 not working if server disabled or down

kevinh-csalabs, 2022-11-23:
## Expected Behavior

IPv6 should still function when server is down

## Current Behavior

IPv6 only works when the server is up/tunnel functioning

## Possible Solution

I'm open to ideas...

## Steps to Reproduce the Problem

  1. Confirm ping -6 google.com works from client and router cli
  2. Disable Server under openmptcprouter > wizard disable server > save and apply
  3. try ping -6 google.com from router cli or client and note that it no longer works
  

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

Kalimeiro, 2022-12-02:
When you disable server, it disable the VPN and IPv6 go through this VPN.

Kalimeiro, 2022-12-12:
> shouldn't IPV6 do the same?

No

kevinh-csalabs, 2022-12-12:
@Kalimeiro Ok, any documented reason why IPV6 would have a different failure path than IPV4? Seems there is at least one bug related to IPV6 handling per https://github.com/Ysurac/openmptcprouter/issues/2687 that has been fixed in dev... I can't find any documented reason why IPV4 would load balance and IPV6 would just stop working. If you can't make IPv6 work with native WAN with IPv6 support then doesn't this prove the problem as it should be working with native WAN?

closed 2023-03-17 by github-actions[bot]