Ysurac/openmptcprouter #2679: Le serveur DHCP ne permet plus d'assigner une passerelle

kobbybest, 2022-11-23:
## Expected Behavior
Le serveur dhcp ne permet plus d'assigner une passerelle de son choix dans les baux statiques

## Current Behavior
La fonction a disparu depuis que je suis passé en V0.59.1

## Steps to Reproduce the Problem

  1.Allez sur la page DHCP puis baux statiques

## Specifications

  - OpenMPTCProuter version: 0.59.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: MILKYWAN
  - OpenMPTCProuter platform: x86_64
  - Country: France

[image]

Ysurac, 2022-11-24:
This will be back in next release.

kobbybest, 2022-11-24:
Ok thanks
Approximately when will the next version be released?
Regards

kobbybest, 2022-11-26:
Est-ce que si je modifie le fichier  /etc/config/dhcp et ajoute  option gw '192.168.1.x', cela va quand même fonctionner ?

Ysurac, 2022-11-26:
oui.
Sinon il faut utiliser le dernier snapshot