Ysurac/openmptcprouter #2680: Does OMR load balance when the VPS is down?

adamb, 2022-11-23:
I'm curious what happens if the VPS fails.   Say I have two connections that I'm bonding. How does OMR handle the traffic in this case?  I could imagine a couple different options.

1.  OMR load balances across all connections 
2. OMR sends all the traffic along the Multipath master.  If the Multipath master fails, it then send the remaining traffic along the secondary path.

If it's number 1,  does it load balance evenly or based on some throughput metric maybe?

Ysurac, 2022-11-24:
By default it load balances across all connections with higher weight on master connection.

Kalimeiro, 2022-11-25:
> By default it load balances across all connections with higher weight on master connection.

but when the master connection is down too, what's connection OMR used, because, it's not working all website try to go through the master down connection. the Glorytun VPN work only.

Kalimeiro, 2022-11-25:
I want to say another important information, when the master connection is down, the VPS appear offline (no token yet available) but it's wrong, the problem is OMR try to communicate with the VPS API through the master connection ... and it is already down ...

I'm in "balanced" mode on the "select master interface" parameter in openmptcprouter advanced settings, so in theorie OMR delete the route on the master connection to the VPS and selecter another high weight connection ? apparently it's not working.

GektorUA, 2022-12-04:
OMR depends of master link, if it's down - you will have down time, while it balance to another link. So - it's no fully failure free.

JRTechServices, 2022-12-09:
Is there a way past this outage while reestablishing with another VPS?

GektorUA, 2022-12-09:
With stock OMR without modifying scripts? I don't think so.

JRTechServices, 2022-12-09:
Thank you GektorUA, if anyone knows how to achieve this I would be happy to back the efforts.

Ysurac, 2022-12-10:
In fact OMR should be failure free and have no down time when master link is down.
Should be a bit better in next release (starting from a fresh config), but there is still case it's not working correctly for now.
With multi VPS, if master VPS fail it should use backup VPS but you will always have a down time as VPS doesn't have same public IP (and proxy/VPN need to reconnect).

JRTechServices, 2022-12-10:
Thank you @Ysurac!

closed 2023-03-15 by github-actions[bot]