Ysurac/openmptcprouter #2683: Shadowsocks is only using one CPU core

NKnusperer, 2022-11-27:
## Expected Behavior
Shadowsocks should use all available CPU cores.

## Current Behavior
Shadowsocks fully utilize one CPU core while the other ones are idling.
This currently caps my throughput to ~160 Mbit.
OpenMPTCProuter is running as a VM using Proxmox.
I first started with 2 cores and then increased the core count to 4 without a differnce.
I notice that only 2 `ss-redir` instances are running, is this expected behavior (the VPS only has 2 cores) or is this something which has been configured at the first boot of the router when the core count was set to 2?

[image]

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: x64

Ysurac, 2022-11-27:
Shadowsocks-libev is not multithread, so at first boot there is a ss-redir created by core.

NKnusperer, 2022-11-27:
Thanks. How can I adjust this after the number of cores has been changed?
Also, what is the expected load pattern on a multi-core system, eg. should they all be utilized equally or is one connection pinned to one core?

Ysurac, 2022-12-01:
You can test chacha20 in wizard, advanced settings checkbox.
If you have good connections, you should really get a better speed than that as we can at least 400Mb/s with a RPI4.

NKnusperer, 2022-12-01:
Unfortunately the result is basically the same using `chacha20`.
After some more testing I think that Shadowsocks is not even the limiting factor here.
When simply using `omr-test-speed` the speed and load is equally high except this time it is caused by curl.
For reference, in a Windows VM I can easily saturate the full 300 Mbit of my LTE connection with moderate CPU usage.

Ysurac, 2022-12-01:
Do you set the VM to use host CPU on the hypervisor ? Check AES is available via ```cat /proc/cpuinfo``` and put back AES encryption in OMR.
Check also you don't have issues in Status->Kernel log and Status->System log.

NKnusperer, 2022-12-01:
Host CPU is in use on the hypervisor and`/proc/cpuinfo` shows `aes` as supported.
No issues observable in kernel/system log.

NKnusperer, 2022-12-07:
> Shadowsocks-libev is not multithread, so at first boot there is a ss-redir created by core.

Can you please point me to the code which is doing this?
On the VPS side this seems to happen here but I was unable to find something on the router side. 
https://github.com/Ysurac/openmptcprouter-vps/blob/44467085a7f9027a87d177a936e496026d0478b0/debian9-x86_64.sh#L661-L664

closed 2023-03-13 by github-actions[bot]