Ysurac/openmptcprouter #2684: OMR 2/3 Links Bonded - 3rd Link Issues

omninet1, 2022-11-27:
## Expected Behavior

Thanks for looking into this 
I have 3 Internet links - 1 Cable 400/10 Mbps, 1 FWA (T-Mobile) 150/25 Mbps & 1 FWA (T-Mobile) 150/25 Mbps -
I should be getting around 700 Down and 60 Up. Other than that everything works great.

## Current Behavior

Instead of getting 700 Down, I get about 400 Down - I get about 60 Up so the upload isn't the issue. 
When I take away one of the T-Mobile links, I get about 550-600, which is what I should be getting.
It's strange that when the 3rd link is enabled, it yields less than 2 connected over both aggregate connections.

All my links are physical links - I run this on VMWare - my VPS is 1Gb dedicated Debian 11. I am getting the speed from the VPS. 
I have changed the encryption to chacha20 (from other threads)
My OMR VM is using VMXNET3 adapters - my links are 1Gbps cooper.

I can change around my wan connections and each will independently get me the above going through the OMR.
If I use Cable and either T-Mobile, results are the same.

Is there some kind of limit on how much speed we can get using OMR?
I plan to use this to deliver 5 Gbps + --- 

The VMs are not resource restricted; one of the VMs is on data-center server hardware w/ 2 vcpu 4 gb and the other is on a Dell workstation (Intel i7) w/ 2vcpu 4 gb

## Specifications

  - OpenMPTCProuter version:

ffrrzz, 2022-11-28:
Both of your T-Mobile cellular router are sharing the same spectrum from the tower back to your home and “fighting” against each other. 

You should be able to use different wireless providers (not resellers of the same service).

omninet1, 2022-11-28:
I appreciate your feedback 
I tested both cellular links independently and together and they are bonding.
I have also switched one of the Tmobile connections with an equivalent Verizon LTE which gets the same speed and the result is the same.
I'm not sure if my physical links are getting overloaded or not?
The connection doesn't seem to get faster than 600-650 (that is with two links on a good day)
Thank you

ccmks, 2022-12-17:
My suspicion is on your VPS.

If you stated that your VPS is 1Gbps, there is a chance, it is only half duplex. That means, the bandwidth is shared 1Gbps between download and upload. Let's say if you are downloading is 600 Mbps, then the remaining 400 Mbps available for upload.

Try to test to see if you can simulate download and upload at same time. I am pretty sure you only getting 500 Mbps when running download and upload simultaneously.

If that's the case, try to get faster VPS, like 2 Gbps above or even 10 Gbps VPS

Keep in mind this OMR is actually VPN on steroid, it relies symmetrical download and upload speed simultaneous connection!

Thank you

closed 2023-03-23 by github-actions[bot]