Ysurac/openmptcprouter #2686: Can I ssh to OMR from the VPS?

adamb, 2022-12-01:
I'm interested in checking out the status of OMR when I'm not at home.  Is there a way to ssh from the VPS to the OMR node running on my Pi?  I can ping 192.68.100.1 but when I try to ssh it says:

```
root@omrlinode:~# ssh root@192.168.100.1
ssh: connect to host 192.168.100.1 port 22: Connection refused
```

This works fine when at I'm at home and want to ssh directly to the OMR machine from the home network.

Kalimeiro, 2022-12-02:
Yes. with port forwarding. See "Documentation" page on OMR Website.

Silvanskii, 2022-12-15:
> Yes. with port forwarding. See "Documentation" page on OMR Website.
I did everything as in the instructions, I don't understand how I can join the web version deleted? which address should I use to join remotely
https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

Silvanskii, 2022-12-15:
its work. i use after ip VPS http://xx.xx.xx.xxx:8000/cgi-bin/luci/

Ysurac, 2022-12-15:
Same way but use another port for VPS side and port 22 for router side.

Silvanskii, 2022-12-15:
> 
Thank you very much, please tell me where I can find out the commands for the device status? to monitor the network connection status of all Eth devices via SSH

Ysurac, 2022-12-15:
You can use ```iftop``` for realtime speed or vnstat for example.

Silvanskii, 2022-12-15:
> 

Tnx! i  install netdata. Also a good solution for monitoring. I meant maybe there is a simple command to control the connection speed. Which can be displayed on the LCD screen

Silvanskii, 2022-12-15:
For information. The netdata works until reboot. After a reboot, it does not work and you need to reinstall

closed 2023-03-21 by github-actions[bot]