Ysurac/openmptcprouter #2687: WAN IPV6 is not being configured properly for use by MPTCP

kevinh-csalabs, 2022-12-02:
## Expected Behavior

IPV6 should utilize MPTCP but currently is not doing so

## Current Behavior

Despite the system and status reporting that IPV6 connection is fine and settings have IPV6 enabled in the openmptcprouter wizard MPTCP is not being used for IPV6 between the router and server. System/kernel logs indicate mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13 which is EACCES: Permission denied. So something is not allowing the kernel to create subflows

I initially pursued this as a problem with MPTCP (see issue 499 link below) but it was determined that something isn't getting configured properly on the openmptcprouter side.

[image]

[image]
Fri Dec  2 14:00:41 2022 kern.err kernel: [1350509.198731] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Fri Dec  2 14:00:43 2022 kern.err kernel: [1350511.428369] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Fri Dec  2 14:00:43 2022 kern.err kernel: [1350511.436115] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Fri Dec  2 14:00:43 2022 kern.err kernel: [1350511.443947] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Fri Dec  2 14:00:43 2022 kern.err kernel: [1350511.452326] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Fri Dec  2 14:00:43 2022 kern.err kernel: [1350511.460512] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Fri Dec  2 14:00:44 2022 kern.err kernel: [1350512.400288] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Fri Dec  2 14:00:44 2022 kern.err kernel: [1350512.408009] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Fri Dec  2 14:00:44 2022 kern.err kernel: [1350512.415897] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13
Fri Dec  2 14:00:44 2022 kern.err kernel: [1350512.423613] __mptcp_init6_subsockets: MPTCP subsocket connect() failed, error -13

https://github.com/multipath-tcp/mptcp/issues/499

## Steps to Reproduce the Problem

  1. Configure using settings wizard and enable IPV6
  2. Set IPv6 assignment length on wan interfaces and restart interfaces
  3. Note IPV6 ping works and system shows IPV6 status as working

## Context (Environment)

Trying to fully utilize IPV6 in addition to IPV4 because IPV4 is much slower and less reliable for 5g/LTE modems for some cellular providers since it is encapsulated

## Specifications

OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
OpenMPTCProuter VPS provider: BuyVM
OpenMPTCProuter platform: RPI 4 64-bit
Country: U.S.

Ysurac, 2022-12-02:
IPv6 WANs are far less tested than IPv4 for now.
I would need the result of ```uci show network```, ```ip -6 a```, ```ip -6 r```, ```ip -6 rule```, ```multipath``` and ```multipath -f```

kevinh-csalabs, 2022-12-02:
network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='5'
network.globals=globals
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_checksum='0'
network.globals.mptcp_subflows='3'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.ula_prefix='fd27:ca97:0230::/48'
network.globals.mptcp_version='0'
network.globals.mptcp_rr_cwnd_limited='Y'
network.globals.mptcp_rr_num_segments='1'
network.globals.multipath='enable'
network.globals.mptcp_scheduler='ecf'
network.globals.mptcp_syn_retries='3'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.congestion='bbr2'
network.globals.packet_steering='1'
network.globals.mptcp_debug='0'
network.lan=interface
network.lan.proto='static'
network.lan.netmask='255.255.255.0'
network.lan.device='eth0'
network.lan.ifname='eth0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.metric='6'
network.lan.label='lan'
network.lan.peerdns='0'
network.lan.ipaddr='192.168.42.1'
network.lan.addlatency='0'
network.lan.ip6assign='60'
network.lan.defaultroute='0'
network.lan.delegate='0'
network.lan.dns='10.255.254.1'
network.lan.ipv6='1'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.omrvpn=interface
network.omrvpn.device='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='100'
network.omrvpn.metric='1200'
network.omrvpn.proto='none'
network.omrvpn.addlatency='0'
network.omrvpn.delegate='0'
network.omrvpn.peerdns='0'
network.omrvpn.dns='10.255.254.1'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.force_link='1'
network.omr6in4.ip6addr='fe80::a00:2/126'
network.omr6in4.gateway='fe80::a00:1/126'
network.omr6in4.ipaddr='10.255.254.2'
network.omr6in4.peeraddr='10.255.254.1'
network.omr6in4.addlatency='0'
network.omr6in4.delegate='0'
[... 293 lines cut ...]
bond0 is deactivated
erspan0 is deactivated
eth0 is deactivated
eth0.2 is in default mode
eth0.3 is in default mode
eth0.4 is in default mode
gre0 is deactivated
gretap0 is deactivated
ifb4eth0.2 is deactivated
ifb4eth0.3 is deactivated
ifb4eth0.4 is deactivated
ip6gre0 is deactivated
ip6tnl0 is deactivated
lo is deactivated
sit0 is deactivated
teql0 is deactivated
tun0 is deactivated
wlan0 is deactivated

Index, Address-ID, Backup, IP-address, if-idx
IPv4, next v4-index: 4
1, 2, 0, 10.191.199.49, 1181
2, 3, 0, 100.96.159.149, 1186
3, 4, 0, 192.168.12.226, 1191
IPv6, next v6-index: 1

Ysurac, 2022-12-02:
You didn't set any IPv6 on WAN here.

kevinh-csalabs, 2022-12-02:
Ugh - sorry about that apparently didn't turn it back on after testing.

Here is the output with option ip6assign set on the wans. Note that option ipv6 gets set to '0' by the router and overwritten if I change it to auto or configure with 1, so if that's the issue it is a bug somewhere.

When I set the ip6assign and the wans get the get an address assigned they briefly appear under multipath -f but then are removed within a few seconds.
 

network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='5'
network.globals=globals
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_checksum='0'
network.globals.mptcp_subflows='3'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.ula_prefix='fd27:ca97:0230::/48'
network.globals.mptcp_version='0'
network.globals.mptcp_rr_cwnd_limited='Y'
network.globals.mptcp_rr_num_segments='1'
network.globals.multipath='enable'
network.globals.mptcp_scheduler='ecf'
network.globals.mptcp_syn_retries='3'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.congestion='bbr2'
network.globals.packet_steering='1'
network.globals.mptcp_debug='0'
network.lan=interface
network.lan.proto='static'
network.lan.netmask='255.255.255.0'
network.lan.device='eth0'
network.lan.ifname='eth0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.metric='6'
network.lan.label='lan'
network.lan.peerdns='0'
network.lan.ipaddr='192.168.42.1'
network.lan.addlatency='0'
network.lan.ip6assign='60'
network.lan.defaultroute='0'
network.lan.delegate='0'
network.lan.dns='10.255.254.1'
network.lan.ipv6='1'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.omrvpn=interface
network.omrvpn.device='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='100'
network.omrvpn.metric='1200'
network.omrvpn.proto='none'
network.omrvpn.addlatency='0'
network.omrvpn.delegate='0'
network.omrvpn.peerdns='0'
network.omrvpn.dns='10.255.254.1'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
[... 141 lines cut ...]
bond0 is deactivated
erspan0 is deactivated
eth0 is deactivated
eth0.2 is in default mode
eth0.3 is in default mode
eth0.4 is in default mode
gre0 is deactivated
gretap0 is deactivated
ifb4eth0.2 is deactivated
ifb4eth0.3 is deactivated
ifb4eth0.4 is deactivated
ip6gre0 is deactivated
ip6tnl0 is deactivated
lo is deactivated
sit0 is deactivated
teql0 is deactivated
tun0 is deactivated
wlan0 is deactivated

Index, Address-ID, Backup, IP-address, if-idx
IPv4, next v4-index: 1
2, 3, 0, 100.96.159.149, 1186
4, 5, 0, 192.168.12.226, 1195
7, 8, 0, 10.191.199.49, 1200
IPv6, next v6-index: 4

Ysurac, 2022-12-02:
I still no see IPv6 address on vlan WANs interfaces.

kevinh-csalabs, 2022-12-02:
Here is what I see from the interfaces view: 
[image]
But IP is local.

Should I be enabling this on wan some other way?

Adding separate interface for dhcp6 isn't ideal and results in "no gateway"
[image]

kevinh-csalabs, 2022-12-03:
So I confirmed setting up a second set of wan connections with dhcp6 the routes show up on mptcp and seem to behave fine, but the connection shows no gateway defined and no ipv6 access which seems to be in error as I can ping ipv6 from the client reliably and it works sporadically via the router (I'm guessing when it brings the interfaces online briefly to check them) so it seems maybe something is wrong with the check...

I'd still rather have a single interface for each connection rather than separate ones for ipv6 and ipv4.

kevinh-csalabs, 2022-12-03:
So debug shows user.notice post-tracking-post-tracking: TMobile6 (eth0.3) switched off because TMobile6 may have ip issues, interface have no IPv4, interface have no IPv4 gateway, interface have no IPv6, interface have no IPv6 

Similar message for all dhcp6 interfaces.

network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='5'
network.globals=globals
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_checksum='0'
network.globals.mptcp_subflows='3'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.ula_prefix='fd27:ca97:0230::/48'
network.globals.mptcp_version='0'
network.globals.mptcp_rr_cwnd_limited='Y'
network.globals.mptcp_rr_num_segments='1'
network.globals.multipath='enable'
network.globals.mptcp_scheduler='ecf'
network.globals.mptcp_syn_retries='3'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.congestion='bbr2'
network.globals.packet_steering='1'
network.globals.mptcp_debug='0'
network.lan=interface
network.lan.proto='static'
network.lan.netmask='255.255.255.0'
network.lan.device='eth0'
network.lan.ifname='eth0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.metric='6'
network.lan.label='lan'
network.lan.peerdns='0'
network.lan.ipaddr='192.168.42.1'
network.lan.addlatency='0'
network.lan.ip6assign='60'
network.lan.defaultroute='0'
network.lan.dns='10.255.254.1'
network.lan.delegate='0'
network.lan.ipv6='1'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.omrvpn=interface
network.omrvpn.device='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='100'
network.omrvpn.metric='1200'
network.omrvpn.proto='none'
network.omrvpn.addlatency='0'
network.omrvpn.delegate='0'
network.omrvpn.peerdns='0'
network.omrvpn.dns='10.255.254.1'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.force_link='1'
network.omr6in4.ip6addr='fe80::a00:2/126'
network.omr6in4.gateway='fe80::a00:1/126'
[... 222 lines cut ...]
erspan0 is deactivated
eth0 is deactivated
eth0.2 is in default mode
eth0.3 is in default mode
eth0.4 is in default mode
gre0 is deactivated
gretap0 is deactivated
ifb4eth0.2 is deactivated
ifb4eth0.3 is deactivated
ifb4eth0.4 is deactivated
ip6gre0 is deactivated
ip6tnl0 is deactivated
lo is deactivated
sit0 is deactivated
teql0 is deactivated
tun0 is deactivated
wlan0 is deactivated

IPv4, next v4-index: 6
0, 1, 0, 192.168.12.226, 375
1, 2, 0, 10.191.199.49, 376
7, 8, 0, 100.96.159.149, 374
IPv6, next v6-index: 8
6, 14, 0, 2605:59c8:6000:2912:e65f:01ff:fea3:0d95, 374
7, 15, 0, 2600:0381:0124:0c03:e65f:01ff:fea3:0d95, 376

Ysurac, 2022-12-04:
There is a bug in omr-tracker, fixed in develop branch.

kevinh-csalabs, 2023-03-06:
any idea when this will make it into a stable build?

closed 2023-06-10 by github-actions[bot]