Ysurac/openmptcprouter #2705: RUTX12    /proc/sys/net/ipv4/ip_forward

highfreq, 2022-12-14:
On my RUTX12 somehow /proc/sys/net/ipv4/ip_forward = 1 is not at defaut anymore. Each time i reboot ip4 is not forwarded anymore. If i run echo 1 > /proc/sys/net/ipv4/ip_forward than it ll starts again untill next reboot.

How do i make it sticky?

Ysurac, 2022-12-15:
Please use template.
Can you check what you have in /etc/sysctld.d/10-default.conf ?

highfreq, 2022-12-15:
I added ip_forward to 10-default.conf yesterday and it worked again. It was empty.

What puzzles me is that i didn't touch the configuration at all in the last month, it just lost ip forwarding on a reboot for some reason.

abenhamo, 2023-03-17:
Hello, I don't open an issue as I didn't purchase the RUTX12. 
@highfreq , could you please tell me if the RUTX12 can handle well at once the aggregation of:
2 x 4G modules + 1 WAN (the one will have a 5G router).
No cpu/stability issues, when to aggregate 3 components ?
Thanks

closed 2023-06-21 by github-actions[bot]

highfreq, 2023-10-18:
I found 10-default.conf empty again after a reboot.
Can't remember what should be in the file. @Ysurac by any chance?

Ysurac, 2023-10-18:
Should be something like this:
```
# Do not edit, changes to this file will be lost on upgrades
# /etc/sysctl.conf can be used to customize sysctl settings

kernel.panic=3
kernel.core_pattern=/tmp/%e.%t.%p.%s.core
fs.suid_dumpable=2

fs.protected_hardlinks=1
fs.protected_symlinks=1

net.core.bpf_jit_enable=1
net.core.bpf_jit_kallsyms=1

net.ipv4.conf.default.arp_ignore=1
net.ipv4.conf.all.arp_ignore=1
net.ipv4.ip_forward=1
net.ipv4.icmp_echo_ignore_broadcasts=1
net.ipv4.icmp_ignore_bogus_error_responses=1
net.ipv4.igmp_max_memberships=100
net.ipv4.tcp_syncookies=1
net.ipv4.tcp_timestamps=1
net.ipv4.tcp_sack=1
net.ipv4.tcp_dsack=1

net.ipv6.conf.default.forwarding=1
net.ipv6.conf.all.forwarding=1
```

highfreq, 2023-10-18:
you mean 10-default.conf should have all that inside? Mine is empty now.

Ysurac, 2023-10-18:
Yes, it the same content on all platform.

highfreq, 2023-10-18:
Ok, thank you very much :)

highfreq, 2023-10-20:
@Ysurac Today after 10-15 reboots, 10-default.conf was empty again. Any idea?

Ysurac, 2023-10-20:
I would say do not reboot :p
My RUTX12 is died due to some I/O failure, I think I killed the flash after many tests...
Check using ```dmesg``` that you don't have any problems like that.

highfreq, 2023-10-20:
something like this:

ath: EEPROM regdomain: 0x0
[   38.058834] ath: EEPROM indicates default country code should be used
[   38.058838] ath: doing EEPROM country->regdmn map search
[   38.058846] ath: country maps to regdmn code: 0x3a
[   38.058852] ath: Country alpha2 being used: US
[   38.058857] ath: Regpair used: 0x3a

highfreq, 2023-10-20:
or:

[   20.526069] UBIFS (ubi0:2): Mounting in unauthenticated mode
[   20.526327] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" started, PID 221
[   20.709226] UBIFS (ubi0:2): recovery needed
[   21.168312] UBIFS (ubi0:2): recovery completed
[   21.168671] UBIFS (ubi0:2): UBIFS: mounted UBI device 0, volume 2, name "rootfs_data"
[   21.171682] UBIFS (ubi0:2): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[   21.179636] UBIFS (ubi0:2): FS size: 75931648 bytes (72 MiB, 598 LEBs), journal size 3809280 bytes (3 MiB, 30 LEBs)
[   21.189522] UBIFS (ubi0:2): reserved for root: 3586439 bytes (3502 KiB)
[   21.199765] UBIFS (ubi0:2): media format: w5/r0 (latest is w5/r0), UUID 7A144764-56F3-4EA7-90BB-C2866329BA58, small LPT model
[   21.216858] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
[   21.255240] block: extroot: not configured
[   21.255593] UBIFS (ubi0:2): un-mount UBI device 0
[   21.258353] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" stops
[   21.278836] UBIFS (ubi0:2): Mounting in unauthenticated mode
[   21.279436] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" started, PID 222
[   21.725401] UBIFS (ubi0:2): UBIFS: mounted UBI device 0, volume 2, name "rootfs_data"
[   21.725489] UBIFS (ubi0:2): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[   21.732403] UBIFS (ubi0:2): FS size: 75931648 bytes (72 MiB, 598 LEBs), journal size 3809280 bytes (3 MiB, 30 LEBs)
[   21.742272] UBIFS (ubi0:2): reserved for root: 3586439 bytes (3502 KiB)
[   21.752551] UBIFS (ubi0:2): media format: w5/r0 (latest is w5/r0), UUID 7A144764-56F3-4EA7-90BB-C2866329BA58, small LPT model

Ysurac, 2023-10-20:
No errors here.

highfreq, 2023-10-20:
I don't see any I/O error in dmesg. It is kind of strange error, it doesn't do it for hundreds boots than all of a sudden it does it 2-3 times in a row. I remember it did it last year too and it was almost new so doubt i/o was misbehaving.