Ysurac/openmptcprouter #2707: Compiling the VPS component on arm an arm64 VPS

rickytrevor, 2022-12-21:
I am trying to compile the MPTCP patched kernel on my Oracle cloud ARM instance

I'm following the official steps to [compile](https://github.com/Ysurac/openmptcprouter-vps/wiki/Compile-kernel) it and it's been stuck ever since  running what I'm guessing is the "build" command
`
fakeroot make-kpkg --initrd --append-to-version=-mptcp kernel-image kernel-headers kernel-source  -j4
exec make kpkg_version=13.018+nmu2 -f /usr/share/kernel-package/ruleset/minimal.mk debian APPEND_TO_VERSION=-mptcp  INITRD=YES 
`

Any clues?

(I am running ubuntu 20.04)

Ysurac, 2022-12-22:
I think it would be better to cross compile for ARM64 support.
Else check CPU/Memory usage after you run the command, and maybe -j4 is too much.

rickytrevor, 2022-12-22:
Any clues on how I could cross compile it?

rickytrevor, 2022-12-23:
Update

I've managed to compile it by installing the aarch64 cross compiler, exporting it and creating the debian package via deb-pkg:

`
CC=/bin/aarch64-linux-gnu-gcc-9
export CC
`

`
fakeroot make -j32 ARCH=arm64 deb-pkg LOCALVERSION=-falcot KDEB_PKGVERSION=$(make kernelversion)-1
`

Another issue accurred tho, my educated guess is that oracle compute instances have secure boot on and, as you can imagine, it won't allow an unsigned kernel to actually boot

I still need to figure that out, since I don't have an arm64 computer at home this is probably going to be a slow debug process, wish me luck

rickytrevor, 2022-12-24:
Another update

I've got my hands on an ARM computer and I tried to run my compiled kernel on a virutal machine there, guess what
The config's broken and it doesn't boot

rickytrevor, 2022-12-24:
Update

I've managed to compile the kernel and boot it off the oracle cloud compute instance, I've also package everything in a neat Dockerfile and a script which do all the heavy lifting, I still need to fix a couple of things but it should be ready in a couple of hours

Now I only need to create compile scripts for everything that the install script pulls from your repo

Ysurac, 2022-12-24:
If you use "SOURCES=yes" option to VPS script, must utils should compile, what is not from source can be disabled and should not be required.

rickytrevor, 2022-12-24:
That's probably going to come in handy, imma keep you updated on everything

rickytrevor, 2022-12-24:
Ok so, it is indeed working as expected, posting this commend trough my oracle compute instance!

Imma take a couple of days off and write down all the instructions to get everything running inside arm vms, don't close the issue yet

rickytrevor, 2022-12-26:
Final update

I've created a small [repository](https://github.com/rickytrevor/openmptcprouter-vps-install-arm64) containing everything that I've used to install openmptcprouter-vps on arm (Compiled .deb files, compile script, compilation Dockerfile and the "custom" installation script, )

Feel free to take the stuff from that repo and merge it with your own, admitted that you want ofc

closed 2022-12-26 by rickytrevor

Network-Traditions, 2022-12-27:
@rickytrevor I was also looking into this and will try to duplicate your setup in my Oracle Cloud Instance.  To date, I was able to create a custom debian 11 image to create the Oracle instance with "Secure Boot" disabled, which results in a bootable version of the OS.  The part I'm trying to work-out now is how I would leverage your [repository](https://github.com/rickytrevor/openmptcprouter-vps-install-arm64) to complete the VPS installation.  Let me know if you have the opportunity to provide some guidance and I'll reply accordingly.  Great work!

rickytrevor, 2022-12-27:
Sure, if you need some help just open an issue **there** and I'll be happy to help