Ysurac/openmptcprouter #2709: Trying to test email

adamb, 2022-12-22:
## Expected Behavior

I have setup the email. I confirm `uci show mail` looks good and no errors in `logread` after `/etc/init.d/msmtp restart`

## Current Behavior

I unplugged one of the wans and I don't get an email.  Is there some place the email sending is logged?  

Is there some way to see when email is sent?  Or where can I see the logs about the wan being dropped or failed over?

## Specifications

  - OpenMPTCProuter version:  0.59.1-5.4
  - OpenMPTCProuter VPS version:  0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider:linode
  - OpenMPTCProuter platform:  RPI4

adamb, 2022-12-22:
basically i'm seeing pretty frequent outages on my starlink. I'm surprised I don't get emails when these happen...

[image]

Ysurac, 2022-12-22:
Did you also check "Mail alert" in Services->OMT-Tracker ?

adamb, 2022-12-23:
> Did you also check "Mail alert" in Services->OMT-Tracker ?

Yes.  Still no emails.  

Is there a way to test the email sending?  Or a place where it's logged?

Ysurac, 2022-12-23:
You can try by using ```echo -e "Subject: test\n\nUn test" | sendmail toemail@domain.com```

adamb, 2022-12-23:
Something isn't right. I checked and the password is correct.  

Here's the error from sendmail:

    sendmail: authentication failed (method PLAIN)
    sendmail: server message: 535-5.7.8 Username and Password not accepted. Learn more at
    sendmail: server message: 535 5.7.8  https://support.google.com/mail/?p=BadCredentials cb20-20020a05622a1f9400b003ab5d6cd6c5sm1557386qtb.16 - gsmtp
    sendmail: could not send mail (account default from /etc/msmtprc)

Here's the `/etc/msmtprc` file

    account default
    host smtp.gmail.com
    port 587
    auth on
    user xxx@gmail.com
    password xxxxxxxxxxx
    from xxx@gmail.com
    tls_trust_file /etc/ssl/certs/ca-certificates.crt
    tls on
    tls_starttls on

If I _only_ select TLS I get this error:

TLS Only config:

    account default
    host smtp.gmail.com
    port 587
    auth on
    user xxx@gmail.com
    password xxxxxxx
    from xxx@gmail.com
    tls_trust_file /etc/ssl/certs/ca-certificates.crt
    tls on
    tls_starttls off

Error:

    sendmail: TLS handshake failed: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
    sendmail: could not send mail (account default from /etc/msmtprc)

Could OMR be using a library gmail smtp doesn't like?

adamb, 2022-12-23:
BTW, I'm trying to find where OMR tracker calls sendmail in the code.  Searching for `sendmail` finds nothing.  Out of curiosity, where is  the code that implements this?

JRTechServices, 2022-12-25:
I have been trying this with O365 myself and have similar results. Let me know if I can help by providing test 0365 smtp credentials.

adamb, 2022-12-27:
I figured it out.  It's a gmail issue.  You need to use App Passwords now to use gmail SMTP.

Here's an article on how to get an app password for this.  

https://forums.raspberrypi.com/viewtopic.php?t=332280

Once I used an App password, this config worked:

    account default
    host smtp.gmail.com
    port 587
    auth on
    user xxx@gmail.com
    password xxx
    from xxx@gmail.com
    tls_trust_file /etc/ssl/certs/ca-certificates.crt
    tls on
    tls_starttls on

Hope this helps!

closed 2022-12-27 by adamb