Ysurac/openmptcprouter #2710: Accept external connections only from specific address

highfreq, 2022-12-23:
@Ysurac i would need a little help to have shorewall on VPS to accept connections for port forwarding only from a specific IP address. Could you please help me out?

Kalimeiro, 2022-12-23:
on OMR you need to go to network -> firewall then edit the rulest, go to advanced settings and add ip source to filter only the ip allowed to connect to this port.

it's not possible to add multiple ip source to one rule, so you need to create a second rule to filter another ip source.

highfreq, 2022-12-23:
Thank you, will try and report back. So nothing needs to be done on the VPS, i can configure on OMR and it'll apply on the server.

highfreq, 2022-12-23:
@Kalimeiro looked a bit around, do you mean inside the  port forward rule, i need to add the filter?

Ysurac, 2022-12-31:
Here:
[image]

highfreq, 2023-01-09:
Found it when kalimeiro pointed it to me, thank you.

highfreq, 2023-01-28:
Hello, would like to know if you think it is possible a porting to the attached hardware. Obviously paying your time.

https://www.microhardcorp.com/DX2.php￼

closed 2023-05-04 by github-actions[bot]