Ysurac/openmptcprouter #2712: Router not giving internet

Roc2707, 2022-12-29:
## Expected Behavior
Wish it would work

## Current Behavior
The server is installed on vmware. The server connects to the VPS, the server's local address is 192.168.88.38. If I set the default IP gateway server address (192.168.88.38), then there is no Internet, as if the server is not a gateway.
[image]
The server works, but if you select it as a gateway, then the Internet does not give.

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64 vmware

Roc2707, 2022-12-29:
If I do a ping for example 8.8.4.4 then it reports that "Timeout has expired"

Kalimeiro, 2023-01-01:
please configure correctly your OMR, your wan settings is wrong/strange/same subnet/...

Ysurac, 2023-01-01:
Why wan1 and wan2 use same gateway ? It's possible and seems to work as you get 2 different public IP. 
You also seems to have all WANs and LAN in same subnet, this is not a good idea.

closed 2023-04-06 by github-actions[bot]