Ysurac/openmptcprouter #2713: Can't get vps token

Polollo, 2022-12-30:
Hi everyone, i'm trying out to set up my openmptcprouter but i cannot go out from an issue. The log keep me say

ri Dec 30 13:29:43 2022 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on "IP", no server API answer on "IP")

i have read about it on github but no solutions works for me... please help me to understand the problem

Ysurac, 2022-12-31:
Please use the template when you open an issue.

The issue is "no server API answer on...", this mean that the port is closed by the host FW or that omr-admin service is not running, in this case make sure you rebooted after VPS script installation and launch ```systemctl restart omr-admin```

Polollo, 2023-01-08:
Hello, happy new year and sorry for late reply, also for not using template
on github. I've tried the command that you say but still didn't work. I
show you my setup to help understand the problem.

Wan 1  192.168.88.1
Wan 2 192.168.1.1
Lan 192.168.100.1

The home server seems to work fine , and i can connect to network using the
wan 1 through Lan port.
I think the problem it's with the vps, hosted on a server that i already
own with centos 7 and a debian 11 machine virtualized on it to run the
server. No problem during installation and reboot, all the port seems to be
open but i still get and error when i try to reach the server. In
attachment some picture to explain better my config.

If you can help me i will  be very grateful to you.

Sun Jan  8 15:28:43 2023 user.warn kernel: [   10.945409] urandom-seed: Seeding with /etc/urandom.seed
Sun Jan  8 15:28:43 2023 user.info kernel: [   10.949867] procd: - early -
Sun Jan  8 15:28:43 2023 user.info kernel: [   11.458689] procd: - ubus -
Sun Jan  8 15:28:43 2023 user.info kernel: [   11.509569] procd: - init -
Sun Jan  8 15:28:43 2023 user.info kernel: [   11.544141] urngd: v1.0.2 started.
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.558470] EXT4-fs (sdd2): re-mounted. Opts: (null)
Sun Jan  8 15:28:43 2023 user.info kernel: [   11.577705] kmodloader: loading kernel modules from /etc/modules.d/*
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.677816] MACsec IEEE 802.1AE
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.679661] natsemi dp8381x driver, version 2.1, Sept 11, 2006
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.679661]   originally by Donald Becker ***@***.***>
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.679661]   2.4.x kernel port by Jeff Garzik, Tjeerd Mulder
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.680570] zram: Added device: zram0
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.683121] NET: Registered protocol family 8
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.683123] NET: Registered protocol family 20
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.685268] NET: Registered protocol family 15
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.685446] Initializing XFRM netlink socket
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.686804] Modular ISDN core version 1.1.29
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.686824] NET: Registered protocol family 34
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.686963] mISDN: Layer-1-over-IP driver Rev. 2.00
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.687055] 0 virtual devices registered
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.687216] DSP module 2.0
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.687218] mISDN_dsp: DSP clocks every 80 samples. This equals 1 jiffies.
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.688758] tun: Universal TUN/TAP device driver, 1.6
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.690167] mISDN: HFC-multi driver 2.03
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.701977] sctp: Hash tables configured (bind 256/256)
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.702756] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
Sun Jan  8 15:28:43 2023 kern.warn kernel: [   11.704921] nat46: module (version 362640b41ae52b732d9e9729e61ac555492442a3) loaded.
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.706343] e1000: Intel(R) PRO/1000 Network Driver - version 7.3.21-k8-NAPI
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.706345] e1000: Copyright (c) 1999-2006 Intel Corporation.
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.707321] igb: Intel(R) Gigabit Ethernet Network Driver - version 5.6.0-k
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.707323] igb: Copyright (c) 2007-2014 Intel Corporation.
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.708099] igbvf: Intel(R) Gigabit Virtual Function Network Driver - version 2.4.0-k
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.708102] igbvf: Copyright (c) 2009 - 2012 Intel Corporation.
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.709623] ixgbe: Intel(R) 10 Gigabit PCI Express Network Driver - version 5.1.0-k
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.709626] ixgbe: Copyright (c) 1999-2016 Intel Corporation.
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.710420] ixgbevf: Intel(R) 10 Gigabit PCI Express Virtual Function Network Driver - version 4.1.0-k
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.710423] ixgbevf: Copyright (c) 2009 - 2018 Intel Corporation.
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.711614] gre: GRE over IPv4 demultiplexor driver
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.711739] ip_gre: GRE over IPv4 tunneling driver
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.712554] ip6_gre: GRE over IPv6 tunneling driver
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.714217] Ethernet Channel Bonding Driver: v3.7.1 (April 27, 2011)
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.721892] i2c /dev entries driver
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.726115] GACT probability on
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.726205] Mirror/redirect action on
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.727035] u32 classifier
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.727036]     input device check on
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.727037]     Actions configured
Sun Jan  8 15:28:43 2023 kern.info kernel: [   11.728334] Simple TC action Loaded
[... cut ...]
Sun Jan  8 15:53:04 2023 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on "ip", no server API answer on "ip")
Sun Jan  8 15:53:16 2023 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on "ip", no server API answer on "ip")
Sun Jan  8 15:53:27 2023 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on "ip", no server API answer on "ip")
Sun Jan  8 15:53:39 2023 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on

Polollo, 2023-01-08:
I reconnect all the cable to the openmptcprouter and update the photo
sended before to not create misunderstanding. The real condition that i can
reach at the moment is this one ( see attachment) . I repeat, i can use bot
connection through the openmptcp router but once at time, still can't
obtain the real bonding passing through the vps

[image: image.png]

closed 2023-04-13 by github-actions[bot]