Ysurac/openmptcprouter #2714: omr-bypass domains ignored for AAAA records

kevinh-csalabs, 2022-12-30:
## Expected Behavior

Adding a domain to omr-bypass should use the specific interface and bypass vps

## Current Behavior

Currently this works if the domain uses A records but AAAA records still go through VPS

## Possible Solution

## Steps to Reproduce the Problem

  1. Add adafruit.com to omr-bypass and set it to use the default master route. enable this rule and save/apply
  2. run traceroute -4 to adafruit.com and note that it uses the master interface
  3. run traceroute -6 to adafruit.com and note that it goes through the vps

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  [v0.59.1-5.4 r0+16594-ce92de8c8c](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version:  1029
  - OpenMPTCProuter VPS provider: buyvm
  - OpenMPTCProuter platform:  RPI4
  - Country:  USA

Kalimeiro, 2023-01-01:
i'm agree, but :

- ideally, you need to have full support IPv6 and IPv4 on each wan.
- mptcp works only with IPv4 OR IPv6, not IPv4/IPv6 mixed .
- currently, IPv6 works only through VPS, it used the VPN.

for mptcp-next, full support IPv4 and IPv6 on mptcp is planned with 6.3 kernel.

for mptcp v0 : i dont know.

Ysurac, 2023-01-01:
I'm working on ignoring AAAA request via option in omr-bypass (all code is done, only need to test and commit)
MPTCP work on IPv4 and IPv6 but in separate way: IPv4 WANs are aggregated and IPv6 WANs are aggregated, not a mix of both, not supported on MPTCP for now and not sure this will ever be supported.

Kalimeiro, 2023-01-02:
> I'm working on ignoring AAAA request via option in omr-bypass (all code is done, only need to test and commit) MPTCP work on IPv4 and IPv6 but in separate way: IPv4 WANs are aggregated and IPv6 WANs [...]

Only with MPTCP-NEXT : 

From [matttbe](https://github.com/matttbe) on his WIP 6.3 kernel :

"
Allow having a mix of v4/v6 subflows for the same socket #269 

It is important to support MPTCP connections with some subflows in v6 and others in v4. But so far, MPTCP does not yet support mixed IPv6 and IPv4 endpoints for the same socket.

Some use-cases:

    Using IPv4 and IPv6 only networks
    Some paths behind different in v4 and v6
    Having only one links with both v4 and v6 where paths to reach the end server are different or using one or the other as a failover mechanism.
    etc.
"

kevinh-csalabs, 2023-01-02:
Thanks - this will be very helpful as I’m finding the ipv4 handling on some cellular networks such as t-mobile is abysmal (much higher latency, packet loss) due to their tunneling.

Regards,
Kevin Hart
(424) 332-7507

closed 2023-04-08 by github-actions[bot]