Ysurac/openmptcprouter #2715: NanoPi R5S / R5C

niklasden, 2023-01-01:
Hi there,

I have a question regarding the use of the new nanopi r5s / r5c (same cpu but c-model has additional wifi module).
There seem to be (unsupported) sysupgrade images available. 

But I cannot figure out how to upgrade the system due to the friendlyarm kernel being version 5.10.xxx, while the images are built for kernel version 5.15 upwards.

Am I missing something there?
Has anyone had success in running openmptcprouter on these devices?

Any help is welcome :)

HonorWater, 2023-01-03:
I've tried differents kernel (5.15,5.4,6.1) from below website:[Generated not tested images (no support on them)](https://snapshots.openmptcprouter.com/)
no lucky here, emmc and sd.
but for openwrt images of 5.15 works fine for me.

niklasden, 2023-01-04:
@HonorWater which OpenWRT images are you referring to? The factory friendlywrt images work fine for me but I have not been able to boot anything custom (e.g vanilla openwrt)

skymedia6300, 2023-01-04:
i am facing  Same Problem  in nanopi R2C

closed 2023-04-10 by github-actions[bot]

kpma1985, 2023-08-26:
i got them booted, on emmc, but the interfaces are lil bit confusing