Ysurac/openmptcprouter #2717: I915  DMC Firmware - best way and right way to install firmware

vinceducat, 2023-01-03:
## Expected Behavior

Solve this issue
[   64.593060] i915 0000:00:02.0: Failed to load DMC firmware i915/skl_dmc_ver1_27.bin. Disabling runtime power management.
[   64.593066] i915 0000:00:02.0: DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
## Current Behavior
have this issue and need to install this firmware
## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: milkywan
  - OpenMPTCProuter platform: x86_64

Ysurac, 2023-01-03:
It's related to graphic card no ?
You can try to download the firmware to /lib/firmware/i915/ maybe this will work.

vinceducat, 2023-01-03:
it copy skl_dmc_ver1_27.bin to /lib/firmware/i915/ but same issue :-(

closed 2023-04-10 by github-actions[bot]