Ysurac/openmptcprouter #2718: WAN (VPS) IP Passthrough

kellenw, 2023-01-04:
Is it possible to pass the public IP from the VPS to a LAN port on my Openmptcprouter? I would like to pass the public routable IP to my Ubiquiti firewall, so I can avoid double NAT as well as allow my firewall to fully manage port forwarding and other routing table rules.

Kalimeiro, 2023-01-05:
No. but you can redirect all ports from the router to your ubiquiti firewall and use full cone nat.

kellenw, 2023-01-05:
> No. but you can redirect all ports from the router to your ubiquiti firewall and use full cone nat.

Hi Kalimeiro,
Thanks for the reply. Yes, I am already redirecting ports from the Openmptcprouter to my Ubiquiti firewall, and that is working great for passing traffic through as intended. I was just curious if there was a way to present the public IP of the VPS host to my Ubiquiti firewall rather than using a NAT'd internal from the Openmptcprouter box.

Steltek, 2023-01-10:
I've asked the same thing in #2711. I can see the external traffic (from the external IP) arrive on my internal system, but the traffic back doesn't seem to make it through OMR unfortunately.

closed 2023-04-17 by github-actions[bot]