Ysurac/openmptcprouter #2720: Some websites unreachable

latranche, 2023-01-07:
Hi,
Since the installation of openmptcprouter I often cannot reach some websites and some others take a very long time to load although my internet connection is fine with a speedtest of around 50Mbs. I use openmptcprouter to agregate Sosh ADSL and Free 4g with a TP-link MR6400 modem.

## Expected Behavior
Access all websites through openmptcprouter

## Current Behavior
Some websites become unreachable sometimes, like mycanal, netflix, strava, amazon or others. Often some of them are reachable but after a very long loading time. And sometimes, all websites are very long to load.
I suspect a DNS issue. I have tried different DNS providers with no change. I have also tried the "steps to solve the problem"  found here : https://github.com/Ysurac/openmptcprouter/issues/41 . It works for around 24hrs then the issue is coming back.

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version:  0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Milkywan
  - OpenMPTCProuter platform: RPI4

Can you help me please ?
You'll find attached my dns configuration in the server settings.

[image]

Ysurac, 2023-01-07:
What is 130.117.11.11 ? here you use 2 DNS servers: 127.0.0.1#5353 that is local OpenMPTCProuter DNS that use root DNS servers, and 130.117.11.11. If one of them doesn't answer this can give long DNS answer time.

latranche, 2023-01-07:
Thanks for this quick answer !
130.117.11.11 is Milkywan DNS. I have removed 127.0.0.1 and 130.117.11.11 and put cloudflare dns 1.1.1.1. I'll tell you tomorrow there is any improvment.

latranche, 2023-01-08:
Hi,
Still having the same problem. Unable to connect to Strava, Brabe search is very long, etc. Encountering this on all my devices (PC, smartphones and Android TV).
Do you have any idea?

Ysurac, 2023-01-08:
Some sites can also ban your VPS IP address. This is why there is omr-bypass.
You can check if it's the problem using ```curl -L www.domainyouwanttotry.com``` from the VPS, if there is no answer than your IP can be banned by this website. Else, you have another problem.

latranche, 2023-01-08:
I don't have access directly to MikyWan VPS, I cannot check with the command. Furthermore, the MilkyWan team told me that there are no banned websites currently.
What is weird is that sometimes one website works and sometimes it doesn't.

Ysurac, 2023-01-08:
True, no problem with MilyWan VPS.
So you have another problem.
IPv6 is disabled ?
Do you have any website that always fails or it's not always same websites ? Do you know if it's a DNS problem or another problem ?
In some case this can be a wrong MTU set on one connection for example, in this case it's often seen as https issues.
Do you have any problem when you try with  curl from OpenMPTCProuter via SSH ?

latranche, 2023-01-14:
Hi, 
IPv6 was enabled.
Strava always fails. Some websites like mycanal, google photos or Android Apps don't work sometimes.
I have tried several things and ended up by disactivating IPv6. And everything works now !
So I guess it was a problem in my IPv6 settings.
However, when I go now to https://test-ipv6.com/ my IPv4 address is my MilkyWan address but the IPv6 one is from orange (I have sosh adsl) and I have a 10/10 score. Is all my traffic still aggregated through openmptcprouter? I thought I wouldn't have any IPv6 address anymore.
Sorry, I am not very familiar with networks....

I guess it is best to have IPv6 enabled, so what do we have to do to enable IPv6 properly? I had just clicked on "enable IPv6" in the advanced parameters and put the gateway's IPv6 addresses in the wizard. I didn't put anything in the "IPv6 address" field as there is no device in the same network as the modem. Maybe this was the cause of my problem?

[image]

Ysurac, 2023-01-14:
If your IPv6 is Orange, it's the problem: you have IPv6 enabled on Orange box and this give you an IPv6, so IPv6 enabled websites doesn't use OpenMPTCProuter. To use correctly OpenMPTCProuter you have to disable IPv6 on Orange box (or IPv6 RA).

latranche, 2023-01-14:
Thanks ! 
So I have disabled IPv6 on my Orange box.
Now, when IPv6 is disabled on Openmptcprouter, everything is fine.
If I enable IPv6, the same problems are coming back (ex: Strava unreachable).
Do we have to do something more than clicking on "enable IPv6" in System->OpenMPTCProuter->Advanced Settings?
(I only did that action).

closed 2023-04-19 by github-actions[bot]