Ysurac/openmptcprouter #2721: Public Subnet on OMR LAN Interface

snejball, 2023-01-07:
Hello there,

I would be in need of some help with configuration:

I have a Public IPv4 /29 Subnet, which I would like to use on OMR LAN Side and be able to route traffic from VPS to it:

The network Is being routed to VPS via a static route, which is working as intended.

However I seem to be missing some configuration, as incoming traffic does not seem to pass from VPS to OMR.
Outgoing traffic from my /29 seems to work fine, however accessing the GUI when configuring the /29 subnet on LAN is no longer possible.

What am I doing wrong?

Ysurac, 2023-01-09:
you have to set route and firewall (/etc/shorewall/*) on VPS side.

closed 2023-04-16 by github-actions[bot]

maegibbons, 2023-06-18:
> you have to set route and firewall (/etc/shorewall/*) on VPS side.

Hi

I am trying to do similar with a /25.

Can you expand on what route you would need to put in on the VPS?

What changes need to be made to the firewall. NOT to nat the vps address and use /25 addresses?

Krs

Mark