Ysurac/openmptcprouter #2725: Connection reset by peer

sonnyxp2008, 2023-01-09:
## Expected Behavior

no connection closed error 

## Current Behavior

the system log is filled with the following error : daemon.err /usr/bin/ss-redir[17625]: remote recv: Connection reset by peer
Mon Jan  9 14:02:38 2023 daemon.err /usr/bin/ss-redir[17626]: remote recv: Connection reset by peer
Mon Jan  9 14:02:38 2023 daemon.err /usr/bin/ss-redir[17626]: remote recv: Connection reset by peer
Mon Jan  9 14:02:38 2023 daemon.err /usr/bin/ss-redir[17626]: remote recv: Connection reset by peer
Mon Jan  9 14:02:39 2023 daemon.err /usr/bin/ss-redir[17626]: remote recv: Connection reset by peer
Mon Jan  9 14:02:40 2023 daemon.err /usr/bin/ss-redir[17626]: remote recv: Connection reset by peer
Mon Jan  9 14:02:41 2023 daemon.err /usr/bin/ss-redir[17628]: remote recv: Connection reset by peer
the ping turns out to be high and I don't understand if the two things are related
Risposta da 8.8.8.8: byte=32 durata=100ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=78ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=99ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=56ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=56ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=44ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=63ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=63ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=397ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=62ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=49ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=330ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=44ms TTL=116
Risposta da 8.8.8.8: byte=32 durata=52ms TTL=116

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: aruba
  - OpenMPTCProuter platform:  x86_64
  - Country:  Italy

Ysurac, 2023-01-09:
Do you have connections problems ?

sonnyxp2008, 2023-01-09:
honestly no, I've tried several tests, including disconnecting one wan at a time, I've tried changing the congestion method but I've had no results

closed 2023-04-16 by github-actions[bot]