Ysurac/openmptcprouter #2728: MULTI LAN PC won't allow wan with dhcp, continuos up/down connection

simonemessina92, 2023-01-10:
- OpenMPTCProuter version: openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined-efi.img.gz
- OpenMPTCProuter platform: x86_64 multi lan rack pc

Hello I've installed OMR into this pc rack with 6 eth ports. As usual I flesh the usb stick with balena etcher to install the os and connect it to the pc. I can easily navigate into the menu of the OMR as I do with other mini pc I've used for this project. But something happen whenever I try to leave active the DHCP on my wans connection, basically the terminal of the OMR and the web gui continue to let me see the wan connection goes up and down. I've tried to settle a manual static IP address and this problem won't happen. Matter of facts, I've tried to not use one of the 5 lan ports but using a USB to LAN adapter leaving the first wan in dhcp , and this problem won't happen either. It seems like whenever a dhcp wan connection is connected to the pc all the ports comunicate through themselves and the omr got crazy, basically It seems like it can enstablish the connection and stay connected but always go up and down. I don't know if there's some particular setting to change in these kind of pc in order to allow every lan port to be on it's own and don't interude into another lan port and cause this problem... Can some one help me with this ? I'm really struggling from several hours right now, I'm pretty sure the pc is working great, it's really reactive with the specs I put on it (i3, 4gb of ram) but this thing it stopped me , I can't even try to use the vps bonding feature if I can't either let the wans work properly one by one.
The system seems to boot correctly at first glance, the first eth port act as the lan port and the other 5 ports act as wans, nothing seems diffrent from another mini pc I was using on daily bases, with the exception I was used that mini pc with one on board lan port and as many USB to LAN adapters as many wan connection I planned to use with it..
I put some shots of the problem.
 Thank you for your help
[image]
[image]
[image]
[image]
[image]
[image]
[image]

simonemessina92, 2023-01-10:
The lan card inside the pc is the following: Controller Gigabit Ethernet Intel® 82583V . I've now tried to install the latest version of openwrt to check if this problem is involved, but indeed it doesn't seems to be. As far as I put my wan1 with dhcp on the eth1 port it worked at first try. I've printed the package list on pdf of the opkg installed into both omr and openwrt. I don't know if there's some specific driver to be installed on OMR to make the intel card works properly, if it is so can some one provide me the right package to install ? I even don't know if this is only driver related issue or not by the way... @Ysurac  can you have a look on this ? thank you so much
[image]
[image]
[omr.pdf](https://github.com/Ysurac/openmptcprouter/files/10385514/omr.pdf)
[open wrt.pdf](https://github.com/Ysurac/openmptcprouter/files/10385516/open.wrt.pdf)

Ysurac, 2023-01-10:
You can try to force link in Network->Interfaces, edit the interface, and "Advanced settings" tab.
As this seems similar to https://github.com/Ysurac/openmptcprouter/issues/2584 check that.

simonemessina92, 2023-01-10:
> You can try to force link in Network->Interfaces, edit the interface, and "Advanced settings" tab. As this seems similar to #2584 check that.

Ok this had fixed the problem of the dhcp on the wan. But due to what I read on the comment under it:
[image]
I've got the subsequential problem that if one connection got unplugged it doesn't seems to work properly. I mean I tried to connect to omr to the vps on the cloud, everything seems fine, but after I hot unplugged the wan1 the bonding won't worked anymore and I can't use the connection, only after a reboot of the pc it had started to work again.... This isn't properly a backup solution I think. I mainly use this system to transmit SRT feed wherever I need multiple connection to be aggregated, it happens  many times that carriers had some problems so I have to swap the external antenna routers to change the isp provider, if the system than stucks it's a major problem. I wonder why do this force link is required while with the normal usb to lan adapter isn't or when on openwrt isn't needed too. Is there some way to have both the two things? all the wans in dhcp and won't loose the connection once one connection need to be replaced while the system is running?

simonemessina92, 2023-01-10:
@Ysurac  What I'm trying to understand is why on openwrt this force link isn't a thing, even if I leve it unthick it works perfectly fine, I mean the pc is still the same one with the same NIC card, so I just think this problem can be overrided somehow on OMR right? Basically the problem accordingly to my setup is that the external wans aren't that much flexible if everytime they had to be swapped a manual reset on the single interface is needed.. I'm far away to get the great effort it took to develop something like that, but at first glance I was afraid my NIC card had some problem, after trying all the thing you suggested me and try openwrt I was absolutely sure that it wasn't the case. Do you think this can be solved diffrently rather than use the force link option? I tried to see if there were some drivers opkg to update on OMR but I didn't found anything about this intel Controller Gigabit Ethernet Intel® 82583V, seems a little bit old, but indeed for some reason on openwrt doesn't have that weird problem on the wans with DHCP on it...

[image]

simonemessina92, 2023-01-11:
@Ysurac I've tried to catch any possible diffrences between openwrt and OMR that might involve this problem, from a driver prospective it seems both omr and openwrt had the same intel drivers installed and they're not require any update so far.
What I've noticed is that the network configuration deamon of openwrt is relatively updated compared to the OMR's one.
I really don't know what implies this fact, I've tried to update that on OMR but after the reboot it seems I can't have any LAN connection anymore, basically it got stucked. So i flash again the usb thumbdrive and charge the latest setting I've done the backup from... 
I'm really getting crazy about why openwrt doesn't have this issue needing of thick forcing the link at all, it works stright away at first boot with the same NIC card, I just want to know if this is something you can try to solve in the next releases or if it is something do we have to deal within. I ask to the pc manifacture if they sell the pc with others NIC cards, but as far as I don't know what it might be the best suitable one I also still need some feedback from you to ask for the correct one, unless of course you don't think this problem can be solved diffrently. The more I test this unit this way the more it seems it doesn't perform the way it should, and that's a pity, because the pc itself it's quite powerful and convenient, don't have to route many external adapters, all the lan ports are ready to work and on the pc itself, I mean I bought this unit to streamline the workflow. Just let me know , best regards.
[image]

Ysurac, 2023-01-11:
I think it's related to igc driver, Linux kernel is not the same on OpenWRT and OpenMPTCProuter. So this will maybe be fixed in next release.
I also added a workaround to set MPTCP needed rules/routes when force link is set.

simonemessina92, 2023-01-11:
> I think it's related to igc driver, Linux kernel is not the same on OpenWRT and OpenMPTCProuter. So this will maybe be fixed in next release. I also added a workaround to set MPTCP needed rules/routes when force link is set.

So I'll wait for the new release ? is this workaround in the release I'm using or do you mean in the next release also? thanks

Ysurac, 2023-01-11:
The workaround will be in next release. You can try snapshots release if you want.

simonemessina92, 2023-01-12:
> The workaround will be in next release. You can try snapshots release if you want.
Where can I download this version ?

simonemessina92, 2023-01-12:
> The workaround will be in next release. You can try snapshots release if you want.

Where I can try to download this snapshots release? I just want to see if it gets better or not, while I'm waiting the new formal release. thanks

simonemessina92, 2023-01-12:
@Ysurac One colleague of mine sent me a flycase rack within a similar machine with 6 lan ports that has exactly the same behaviour, dhcp loop, that is solved only within using the force link. I'm guessing this mount the exact same NIC chipset of my rack pc at this point. I'm not sure 100% because I can't have any manual of it, is there a way to check what is the model of the NIC card through the terminal? So just in case I can also feedback if it's another type of card or is the same I've got in the other pc... we both bought these pc units from China, so presumably it mounts the same NIC, but if there some comand please let me know so I can inform you to better troubleshoot this. Thank you !
[image]
[image]

simonemessina92, 2023-01-12:
@Ysurac I'm trying some more testing, and I've seen that no matter what this force link option is mandatory somehow with these Pcs and NIC cards... By creating a wan standalone with static Ip address I see that this option is tick as standard option, and no matter what if I leave this untick it doesn't work, there's always the dhcp loop on the interface.
By the way no matter what I try this OMR won't work good with these machines, I can't believe I've got 5 LTE routers cat6 and got so bad results (see the attachment). Every single LTE router goes about 70mbps download / 30mbps upload in standalone mode connected to a device.

[image]

 With the other mini pc I have where I use the USB to LAN adapters, that won't have the necessity to use the force link option the performance are actually great... 

I've got also the problem in the OMR interface that says "TCP: too many orphaned sockets", and subsequential "MPTCP: out of memory: force closing sockets"

[image]

I've seen there's also another post #2729 about this problem... I really can't make these 2 pc works... Please I really need some help with it, got an events coming where I don't have any wire internet line, and without this working I don't know what to do...

As I said last test was also made all with static IP addresses per each wan, so I really don't understand how this can be solved if even this way doesn't seems to work properly.

Ysurac, 2023-01-12:
Again, did you try snapshots images ?
Link is on download page : https://snapshots.openmptcprouter.com/

simonemessina92, 2023-01-12:
> Again, did you try snapshots images ? Link is on download page : https://snapshots.openmptcprouter.com/

Do you intend  the new alpha on the 6.1 folder this one ? If it is like so , no I didn't still tried that, I'm now testing the old mini pc I was using everytime with usb adapters to see if even that had some sort of issue like this or not. Next step is to try the latest release I guess. Do I have to change something in the VPS to use that OMR release or I can leave the server the way it is? Thank you.
[image]

simonemessina92, 2023-01-14:
@Ysurac sorry for the late reply but I got intense tests on both the two machines, the rack one and the mini pc too. It seems that running this 0.59.2 alpha the diffrence is noticible. I was able to run all the wans interfaces in DHCP, still there's the problem I have to use the force link option ( which still in this version is only needed with the actual NIC card on the pcs not when using the USB to lan adapters); but performance wise it's a diffrent story.
Everything seems to start at boot normally, connection to the vps requires as long as 1minute almost, but after the connection got established the OMR works fine. I don't see that TCP socket error, actually I've seen that only one time at the first configuration on both pc, but after the first reboot everything seems gone back to normal usage.
So as far as it appears from my tests this version works quite better than the current 0.59.1 at least with these kind of machine with multiple LANs on it, the old mini pc with usb to lan adapters worked fine even with the 0.59.1...
Btw I really think I won't touch anything until the new version will be stable , as long as it seems to work fine at least with this one.
Still wondering if this force link is really necessary, I mean at the moment yes because even in the 0.59.2 I got dhcp loop on the internal lan ports if I don't leave ticked that (both dhcp,static mode), but if I attach usb to lan adapterds, doesn't seems to be a necessity, it's just work with those peripherals.
So i patiently wait for the next stable release to see if the situation will be even better than now. I much appreciate your suggestion on this problem, thank you again for everything!

closed 2023-04-19 by github-actions[bot]

spritacco, 2023-06-16:
Hi, Was a fix ever discovered for this problem? I've just run into it.

simonemessina92, 2023-06-16:
> Hi, Was a fix ever discovered for this problem? I've just run into it.

I don't really think so, I would like to try a new version which solve this kind of issue, I think it's a quite important thing to solve afterall, because everytime i change a WAN I have to remember to reconnect the interface otherwise the thing simply don't work. And the fact that the force link should be enabled It's not quite convenient.

khdegraaf, 2023-10-20:
I believe the ultimate cause of this issue is outlined in https://github.com/Ysurac/openmptcprouter/issues/3005. I found a fix/work-around for anyone interested in this.

simonemessina92, 2023-11-24:
Hello @Ysurac  can you please provvide me the exact link on the version 0.59alpha2 I can't find it anywhere now, I just see the new 0.60 version. Furthermore, I was looking at #3005 and I can see there's a file to modify to avoid the 2.5nic interfaces to continue loosing the dhcp lease, how to do exactly this? 
Furthermore, is there any guide to install OMR properly onto a VM like Virtualbox for example and route the lan of OMR directly to the VirtualBox Adapter on the Windows Pc without causing issues on the Windows ? I would like to test this system running directly in a VM and give bonded internet to the windows machine without using 2 separate hardwares, but I didn't get if it is possible or  should I required something different than virtualbox, like Proxmox and similar hypervisors... thank you