Ysurac/openmptcprouter #2729: MPTCP Out of memory and too many sockets

ByteArrayException, 2023-01-11:
## Expected Behavior
No error logs and good working.

## Current Behavior
The kernel log has a lot of entries like this after Update from v0.58.5 to v0.59.1:
[261081.351086] TCP: too many orphaned sockets
[261081.351093] TCP: too many orphaned sockets
[261081.351117] TCP: too many orphaned sockets
[261081.351124] TCP: too many orphaned sockets
[261081.351126] MPTCP: out of memory: force closing socket
[261081.351154] TCP: too many orphaned sockets
[261081.351173] TCP: too many orphaned sockets
[261081.351182] TCP: too many orphaned sockets
[261081.351191] TCP: too many orphaned sockets
[261081.351193] MPTCP: out of memory: force closing socket
[261096.028928] net_ratelimit: 273 callbacks suppressed
[261096.028929] TCP: too many orphaned sockets
[261112.003613] TCP: too many orphaned sockets
What does this entries mean?
The client/omr has 16GB of RAM and no load on it, so there is no bottleneck.

Thanks for your help!

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: x86_64

Ysurac, 2023-01-11:
You can try to comment line ```net.ipv4.tcp_max_orphans```, ```net.ipv4.tcp_mem```,```net.ipv4.tcp_rmem``` and ```net.ipv4.tcp_wmem``` in /etc/sysctl.d/shadowsocks.conf and reboot.

ByteArrayException, 2023-01-15:
Thanks for your help!

closed 2023-01-15 by ByteArrayException