Ysurac/openmptcprouter #2730: Raspberry PI Compute 4 I/O board and CM4

bluisana, 2023-01-12:
Hello, 

I was pumped to tryout openmptcprouter on my new raspberrypi setup.  Unfortunately none of the images for CM4 will boot on my board :(.  Are there any images that use a newer PI firmware version hidden somewhere?  

[image]

Ysurac, 2023-01-12:
You can try snapshots images

bluisana, 2023-01-12:
Thanks Ysurac,

bluisana, 2023-01-13:
Hey Ysurac, 

I don't know if this information would help you or not if you wanted to support this board but the latest image of openwrt worked fine for me. 

https://downloads.openwrt.org/releases/22.03.3/targets/bcm27xx/bcm2711/openwrt-22.03.3-bcm27xx-bcm2711-rpi-4-squashfs-factory.img.gz

bluisana, 2023-01-15:
Hey Ysurac, 

Is it possible for me to try to build an image of openmptcprouter for the version of openwrt that works on my device?  

If so could you give me a basic run down of how to accomplish this? 

Thanks,

Ysurac, 2023-01-18:
You can try latest 5.4 snapshots on https://snapshots.openmptcprouter.com/

bluisana, 2023-01-19:
Hey Ysurac, 

I still get the start4.elf: is not compatible.  This board requires newer software message with the v0.59.2alpha-5.4-r0+16819-1b6e9b3f64-bcm27xx-bcm2711-rpi-4-ext4-factory.img.  I will just keep checking back here periodically to look at the snapshot images.  If you remember, shoot me a message on here when you get the rpi4/bcm27xx images for 6.1 uploaded and I will give it a shot.  

Thanks

closed 2023-04-24 by github-actions[bot]

bluisana, 2023-06-22:
I booted up with the latest raspberry pi os and it shows the driver for the built-in eth0 to be bcmgenet.

bluisana, 2023-06-22:
I believe this is related to https://github.com/openwrt/openwrt/issues/11608 but I have no way to resolve it :(.