Ysurac/openmptcprouter #2731: Low bonding performance when one or more interfaces are low on BW

kmrinal, 2023-01-15:
Hi, We have been testing the OMR for a while now, where we use OMR to bond 4 different 4G/LTE WAN connections. During our tests, we observed that when one/two of the interfaces are low on bandwidth it tanks the overall bonding upload and download performance. Like in one of our tests, we saw this

  | SIM1(without OMR) | SIM2(without OMR) | SIM3(without OMR) | SIM4(without OMR) | Bonding Performance (OMR)
-- | -- | -- | -- | -- | --
Download | 7.18 Mbps | 13.32 Mbps | 6.37 Mbps | 13.72 Mbps | 23.57 Mbps
Upload | 1.81 Mbps | 22.28 Mbps | 2.53 Mbps | 15.19 Mbps| 13.65 Mbps
  |   |   |   |   |  

Presently we are planning to disable OMR participation for the least-performing interfaces (if it is below a threshold). Wondering if there an elegant/better way to avoid degraded bonding performance?

Ysurac, 2023-01-15:
Please use the template, it's not possible to help without versions, platform,...

kmrinal, 2023-01-16:
Sure @Ysurac , will repost it. Closing the issue here. Thanks.

closed 2023-01-16 by kmrinal