Ysurac/openmptcprouter #2732: Internet stopped working over usb android/iphone tethering after short time - also directly on phone

zbynekdrlik, 2023-01-15:
## Expected Behavior
Stable aggregated connection over multiple internet wired/usb tethered connections. 

## Current Behavior
OMR is installed on Intel NUC computer. Lan is connected through eth0. 
Internet is connected over usbeth0 convertor and usb android or Iphone connections.
We are dealing with speed-down connections over phone connection to very weird state. When it is happened also speedtest
run directly on phone stopped working. If we run some ping app it is working correctly. 
Shortly after usb cable is disconnected from phone, speedtest start working on phone.
Sometimes it works few hours and sometimes it goes to this state in  minutes.
When connection is corrupted, openmptcprouter wizard state still show that connections is green.
We also examine logs but in minutes when this issue appear there is no change in syslog/dmesg...

We also try to change our vps router ip but, without help. 
Test multiple sim cards and operators, same behavior.

## Possible Solution
Find the fix

## Steps to Reproduce the Problem

  1. Connect android phone by usb with usb tethering
  2. Use till issue appear, in that case, internet stop working also directly on phone.
  

## Context (Environment)
We are trying to have backup of our internet connection with help of usb tethering.

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4-r0+16594
  - OpenMPTCProuter VPS version: v0.59.1-5.4-r0+16594
  - OpenMPTCProuter VPS provider: websupport.sk/contabo/terrahost/linode - tested multiple but we lost tracking point, because we do it to find the bug
  - OpenMPTCProuter platform: x86_64
  - Country: Slovakia

Ysurac, 2023-01-15:
It's happening only on speed test or after a delay ? Always same delay ? Do you have something in system or kernel log when you have the issue ?

zbynekdrlik, 2023-01-15:
I have still not yet fond exact relation to some time or some other behavior. Few time it was working excellently, but now it not working for after very short time. I have been scanning dmesg -c and logread -f to see if something is reported there in that time. But not found anything.

zbynekdrlik, 2023-01-23:
Hi, I i have moved forward in this, for some more tests, the situation is that mptcp data is still running through phone in redundant mode but browsing directly on phone is not working. Same situation on android on iphone.

closed 2023-04-28 by github-actions[bot]