Ysurac/openmptcprouter #2734: Run MPTCPRouter in LXC container

sinlam, 2023-01-16:
Hi all,
  Is it possible to run MPTCPRouter in a LXC container or LXC VM (not using Proxmox)? My LXC is installed in a Ubuntu 22.04 x86_64 machine. I use OpenMPTCPRouter rootfs file (openmptcprouter-v0.59.1-5.15-r0+20029-3c06a344e9-x86-64-generic-rootfs.tar.gz) and created a metadata.yaml file, then run 'lxc image import <meta tar> <rootfs gz> --alias mptcprouter'. Image import is successful, and execute 'lxc start mptcprouter'. The container is running, but I cannot access MPTCPRouter web UI, no IP is allocated from the router. 

Anyone know how to make it work? Access to the container shell but I don't know how to debug. 

  - OpenMPTCProuter version: 0.59.1-5.15
  - OpenMPTCProuter VPS version: kernel 5.15
  - OpenMPTCProuter VPS provider: internal machine
  - OpenMPTCProuter platform: x86_64

Thanks for any suggestion!

Ysurac, 2023-01-18:
To use LXC, docker,... the host kernel must have MPTCP support enabled.
Else you can check OpenWRT documentation for LXC, maybe this can help.

sinlam, 2023-01-18:
Hi Ysurac, thanks for your reply. Ubuntu 22.04 already has MPTCP support enabled in the kernel. I created two software bridges and integrate into the MptcpRouter container in LXC. Login to the MptcpRouter container and I can see the 192.168.100.1 IP address, but it cannot allocate new DHCP IP address to connected host. Is there anything i can check?

I will read OpenWRT document for LXC, thanks for the suggestion!

closed 2023-04-23 by github-actions[bot]