Ysurac/openmptcprouter #2735: L2TP

EZPC-Matt, 2023-01-16:
I have an plain L2TP tunnel running from behind openmptcprouter to an external LNS (Over Starlink + VDSL) and other than the latency spiking quite high when the links are congested it seems to work quite well thus far (I need to do some more testing to be sure), it seems when the traffic levels are low the DSL gets prefered (presumably due to it's lower RTT time) but you get the additional bandwith Starlink provides when needed.

This makes me wonder if L2TP would be useful as a tunnel type as an alternative to Glorytun.etc

Ysurac, 2023-01-18:
For OpenMPTCProuter we need tunnel encryption so L2TP+IPSEC dans this use UDP. So can be usable to replace OpenVPN or Wireguard for MPTCP over VPN, but not to replace Glorytun.

EZPC-Matt, 2023-01-18:
If you used IPSEC it would be proto ESP + UDP on probably either 500 or 4500 so I doubt that would play nice with MPTCP.
Why do we need tunnel encryption?

In some scenarios yes I can see it being needed but not in all, also presumably the L2TP is transversing ShadowSocks/V2RAY which would be encrypting it (Unless you've turned that off)

ccmks, 2023-01-26:
> If you used IPSEC it would be proto ESP + UDP on probably either 500 or 4500 so I doubt that would play nice with MPTCP. Why do we need tunnel encryption?
>
> In some scenarios yes I can see it being [...]

I believe because it only encrypts from OMR to VPS, but from VPS onward won't be encrypted if not using IPSec

Ysurac, 2023-01-26:
Because using a tunnel over a Proxy will be slower than using directly the proxy. All VPNs connect directly to the VPS. And all traffic between router and VPS is encrypted by default.