Ysurac/openmptcprouter #2737: Random interrupts

ByteArrayException, 2023-01-21:
## Expected Behavior
Smooth MPTCP Usage

## Current Behavior
Since the update from 0.58.5 to 0.59.1 the internet connection is always disconnected randomly after some time.
In the log then appears:
Fri Jan 20 22:00:31 2023 daemon.err /usr/bin/ss-redir[22590]: send: Broken pipe
Fri Jan 20 22:00:33 2023 daemon.err /usr/bin/ss-redir[22591]: send: Broken pipe
Fri Jan 20 22:00:38 2023 daemon.err /usr/bin/ss-redir[22592]: send: Broken pipe
Fri Jan 20 22:00:40 2023 daemon.err /usr/bin/ss-redir[22590]: send: Broken pipe
Fri Jan 20 22:00:40 2023 daemon.err /usr/bin/ss-redir[22591]: send: Broken pipe
Fri Jan 20 22:00:40 2023 daemon.err /usr/bin/ss-redir[22593]: send: Broken pipe
Fri Jan 20 22:00:46 2023 daemon.err /usr/bin/ss-redir[22590]: send: Broken pipe

## Specifications

- OpenMPTCProuter version: v0.59.1-5.4
- OpenMPTCProuter VPS version: 0.1028
- OpenMPTCProuter VPS provider: Hetzner
- OpenMPTCProuter platform: x86_64

Ysurac, 2023-01-21:
There is no enough info to do anything...
What do you have in status page ? What you mean by internet connection ? ping, webpage, connection to VPS,... ?
Do you have any other log before/after this (the log is related to shadowsocks) ?
This happen when you have high usage or not ?

ByteArrayException, 2023-01-21:
Nothing appears in the status page, everything is still green and online. The complete internet connection is gone, so UDP, TCP, ICMP all together. This is from the syslog, here is an extended excerpt: 
Fri Jan 20 21:53:03 2023 authpriv.info dropbear[32329]: Exit (root) from <192.168.100.207:50181>: Exited normally
Fri Jan 20 22:00:31 2023 daemon.err /usr/bin/ss-redir[22590]: send: Broken pipe
Fri Jan 20 22:00:33 2023 daemon.err /usr/bin/ss-redir[22591]: send: Broken pipe
Fri Jan 20 22:00:38 2023 daemon.err /usr/bin/ss-redir[22592]: send: Broken pipe
Fri Jan 20 22:00:40 2023 daemon.err /usr/bin/ss-redir[22590]: send: Broken pipe
Fri Jan 20 22:00:40 2023 daemon.err /usr/bin/ss-redir[22591]: send: Broken pipe
Fri Jan 20 22:00:40 2023 daemon.err /usr/bin/ss-redir[22593]: send: Broken pipe
Fri Jan 20 22:00:46 2023 daemon.err /usr/bin/ss-redir[22590]: send: Broken pipe
Fri Jan 20 22:01:47 2023 daemon.err /usr/bin/ss-redir[22593]: send: Broken pipe
Fri Jan 20 22:05:48 2023 user.notice post-tracking-post-tracking: Check API configuration...
Fri Jan 20 22:05:51 2023 user.notice post-tracking-post-tracking: Check API configuration... Done

There was no high load on any connection, it often occurs in the middle of browsing the web. Before the update everything was working fine. Except that MLVPN doesn't seem to work since the update. In the status page it says permanently "VPN not running", is there by any chance a UDP speed test to check if it is really down?

Ysurac, 2023-01-21:
Why do you use MLVPN ?
To check if it's really down, use SSH on router part and try to ping 10.255.253.1 (the VPS MLVPN endpoint)

ByteArrayException, 2023-01-21:
I use MLVPN for UDP Aggregation (cause that gave me the best result lately, V2Ray bonding well, but I have to fight with many disconnects after a connection time of more than 24 hours, Shadowsocks does this 1x per day, v2ray however several times per day after a connection time > 24 hours)

winkmichael, 2023-01-22:
> Why do you use MLVPN ? To check if it's really down, use SSH on router part and try to ping 10.255.253.1 (the VPS MLVPN endpoint)

Whats wrong with MLVPN? That what I've been using with great success and performance for months.