Ysurac/openmptcprouter #2745: Site to Site VPN

ccmks, 2023-01-29:
Hello

Is this possible to setup site to site VPN in OMR environment?

For example, both site A and site B has 2 or 3 different internet connections. I would like to aggregate all internet connection on both site to establish site to site VPN

I kind of figure if I am using IPsec or L2TP tunnel over OMR, the speed will be drastically slow to the point that not using OMR will have better speed.

I like the idea of using OMR to get out to internet, but I would like to see how we can implement this in site to site environment

Thank you

Kalimeiro, 2023-01-29:
Yes with OpenVPN integrated in OMR.

I use it for a VPN site-to-site and it works great but need a good knownledge on network and how to configure correctly openvpn.

ccmks, 2023-02-03:
You used OpenVPN for site-to-site in OMR to get aggregation?

closed 2023-05-10 by github-actions[bot]