Ysurac/openmptcprouter #2747: OpenMPTCProuter v0.59.2 alpha

Ysurac, 2023-01-29:
OpenMPTCProuter v0.59.2 alpha snapshot release is available with 3 Linux kernel : 5.4, 5.15 and 6.1.

5.4 is the "stable" branch with fullmesh, redundancy,... It use the out of tree MPTCP release.
5.15 and 6.1 kernel only support fullmesh for now, so can only aggregate all connections. It use MPTCP upstream version. Only a few platforms are available on 6.1 for now.

All release are available on https://snapshots.openmptcprouter.com/

Detailed changelog will be available here soon.

It's an alpha release, so can have many bug and some features are missing for final release.

NKnusperer, 2023-02-07:
Can you please elaborate what `only support fullmesh` means practical speaking?
eg. losing the master connection is not supportend or bypass rules are not working?
Looking forward for v6.1 kernel because of hardware support.

JRTechServices, 2023-02-07:
Thank you Ysurac for all of your hard work.

fareign, 2023-03-09:
[image]
I sovled vnstat2 cant display image in 5.15 kernel.but I dont know why.

1. remove vnstat2
2. install vnstat
3. remove vnstat
4. install vnstat2 then is ok.

maybe some conflict with vnstat.

Steltek, 2023-06-05:
Tried 5.15, but couldn't install keepalived due to missing dependencies. (Without it my other router takes over.)

metro2450, 2023-06-18:
Hi Ysurac,

Thank you for the wonderful project.
May I know if there is a plan to move the project base to the new openwrt 23.05 release with 5.15 kernel and provide more device support since MPTCPv1 is supported in 5.15 kernel out of box?
https://openwrt.org/releases/23.05/

Ysurac, 2023-06-18:
Not really, but there is plan to move to latest OpenWRT and use 6.1 kernel. Most arch are already supported on develop branch.
On next release 6.1 will be official release and 5.4 for legacy release (and unsupported arch on 6.1)
5.15 doesn't give a really good result.

vempire-ghost, 2023-06-24:
In the last snapshot from de v0.59.2 alpha the proxy is not showing when v2ray is selected and the system log is always blank.
[image]
[image]

hle5128, 2023-08-27:
i notce that you haev 6.1 target Ipq807x chip, is that mean I can test the gl slate Issue: https://github.com/Ysurac/openmptcprouter/issues/2415

Ysurac, 2023-08-27:
Support was tested with qnap-301w (platform is qualcommax), GL-AXT1800 is still not supported by OpenWRT and I don't have the device so I can't add/test support.

hle5128, 2023-08-27:
> Support was tested with qnap-301w (platform is qualcommax), GL-AXT1800 is still not supported by OpenWRT and I don't have the device so I can't add/test support.

Have you check here? It supported here:

https://forum.gl-inet.com/t/gl-inet-ax1800-openwrt-firmware/19447/257

Ysurac, 2023-08-27:
There is no official support. So it's not always easy to add without the device as I may have to changes somes patches, change kernel config,...

closed 2023-11-30 by github-actions[bot]