Ysurac/openmptcprouter #2748: No Aggregation / slow speed

psylou, 2023-01-29:
## Expected Behavior

2 WAN's should get aggregated with about 1000Mbit/s

## Current Behavior

WAN1 uses almost full speed but WAN2 about 5% and I get about 20 Mbps
WAN1 is 250/40
WAN2 is 750/40
Router has 4 Cores / 4 GB
VPS has 4 Cores / 8 GB

[image]

omr-test-speed
[image]

[image]

omr-test-speed eth1 (WAN2)
[image]

omr-test-speed eth2 (WAN1)
[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59.1-5.15  
  - OpenMPTCProuter VPS version: openmptcprouter v0.59.1-5.15 
  - OpenMPTCProuter VPS provider: 1blu
  - OpenMPTCProuter platform: x86_64 Proxmox VirtIO NICs

Ysurac, 2023-01-30:
5.15 based release is still an alpha version.
omr-test-speed should be run at least 2 minutes to get a real result.

psylou, 2023-01-31:
Thank you for replying. But I need the upstream aggregation. Actually I switched to your 6.1 alpha release and it seems to be working there. My interfaces have very different latencys. Any way to let VoIP or games go over one interface but still use the vpn?

closed 2023-05-07 by github-actions[bot]

AyberkAltuntop, 2024-03-02:
Hello,
The same applies to me, I connect 3 wans, 1.000/100, 80/8 and one variable speed cellular connection. In my tests I could not exceed 250 Mbps download 100 upload. 

Where should I look? 

When I test the VPS and wans one by one, it gives the values that should be.

Ysurac, 2024-03-02:
@AyberkAltuntop please open a new issue, and if you use 5.15 release: don't use it.

AyberkAltuntop, 2024-03-02:
Thank you for your answer. I reinstalled my VPS as kernel 6.1 with Debian 12. I ordered Qnap Qhora 301-w for the router. 

I will test it again during the week and if my problem persists I will open a new topic. 

By the way, this is a very nice and successful job, thank you for giving it to us.