Ysurac/openmptcprouter #2749: Multiple IPs on VPS not working as expected

NiColtro, 2023-01-29:
## Expected Behavior
Following instructions on wiki page https://github.com/Ysurac/openmptcprouter/wiki/VPS-Multi-IPs, the router should use the configured public IP for the specified device.

## Current Behavior
All status flags in openmptcprouter status page are green.
In services - shadowsocks-libev
- redir rule oip2_rule has been setted up following provided documentation (enabled + src ip/net forward)
- remote server has been enabled (oip2server)
- local istance is now in running (ss-redir.oip2)

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4
  - OpenMPTCProuter VPS version: v0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: OVH with 2 additional single IPv4 addresses
  - OpenMPTCProuter platform: x86_64 - Debian 11

Syslogs: https://hastebin.com/epaxaqudoh.yaml

Ysurac, 2023-01-30:
I'm missing what you have done and what you want.

NiColtro, 2023-01-30:
Hi Ysurac, I'm aware of the fact that what i gave you are very poor informations about what i want to achieve, but I don't really know what I can provide you to take this further.
As I said, I deployed a Debian 11 VPS istance on OVH and runned the provided installation script.
Previously, the 2 needed additional public IPs have been configured and tested on the machine itself (so the total of IP available is 3).

After that, I installed and configured the OMR on a x86 based system.
Following the documentation provided in the wiki (https://github.com/Ysurac/openmptcprouter/wiki/VPS-Multi-IPs), the oip redirect rules (on Shadowsocks config) have been set up with the local LAN ip address of the end device in the forward src list.
Done that, I enabled the related remote server and local instance of ss. At the end of the configuration process the OMR is not making the specified device reach Internet with the expected public IP address.

Let me know what informations I can provide to you to solve this.
Many thanks in advance.

RoFiTech, 2023-02-01:
Hello there!

I am in exact the same boat. I have tried everything i mentioned that can help, but only the primary IP from the VPS is used.

philmor90, 2023-02-06:
Exact same Problem here!

NiColtro, 2023-02-09:
Hello, everyone.
Has anyone managed to solve the problem?

Ysurac, 2023-02-11:
@NiColtro I would need the result of  ```netstat -laputen | grep ss-server``` from the VPS.
I think the issue is on VPS side, I will fix that.