Ysurac/openmptcprouter #2751: OMR Bypass is broken?

darthclide, 2023-01-31:
## Expected Behavior
OMR bypass should take any port or url and send it on the chosen interface.

## Current Behavior
As soon as I enable omr bypass for even a simple url like reddit.com it then blocks the page. Does not even reroute it to my chosen interface. As soon as I disable that rule, the site loads just fine.

## Possible Solution
Looking through other bug reports I saw something about AAAA records, but I am unsure how this would be my problem since it worked decently enough before this update.

## Steps to Reproduce the Problem

  1. Add OMR bypass rule
  2. Try to load the website in that rule.

## Context (Environment)
Even in the past versions OMR-bypass has been flaky at times. But I would at least hope it wouldn't straight up block the connection when you choose an interface in OMR-bypass.

## Specifications

  - OpenMPTCProuter version: 59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Vultr
  - OpenMPTCProuter platform: RPI4
  - Country: USA

darthclide, 2023-01-31:
Okay now I am really confused... I just disabled all of my domains in OMR-Bypass and yet all those urls were still blocked. But then when I disabled all of my port rules, it then allowed those urls again? It is almost like sometimes OMR-Bypass is not applying the rules even when you hit "save & apply".

And no, I did try loading other web pages during my testing and anything NOT on the OMR-Bypass list worked just fine.

darthclide, 2023-01-31:
I have IPv6 disabled on my computer, but I see IPv6 is enabled on the interface I was trying to bypass to. But that interface is my local ISP who is not using IPv6 (and no plans to), so that shouldn't matter right?

Is there a way to reset the cache without restarting? A few of my hotspots require me to manually re-enable after rebooting so I prefer avoiding that if possible.

darthclide, 2023-01-31:
I got up today with all the urls working and decided to try omr-bypass with one one url lichess.org and it worked. But then when I tried to activate the rest of my 30 rules, all the urls are blocked again (including lichess.org)

(I did disable IPv6 on that interface even though I doubt that is the problem here)

Ysurac, 2023-02-01:
You may have a problem in one of your URL.

darthclide, 2023-02-01:
How is reddit.com a problem? Or lichess.org

I didn't have problems before 59.1-5.4 but as I mentioned OMR-bypass has always been finicky since I started using this software 2 years ago. It feels like sometimes the rules are not applied, or they are applied, but randomly traffic somehow ignores those rules.

Is there a way to reset DNS without rebooting?

Ysurac, 2023-02-02:
If you don't want to try anything, no need for this issue. I close it.

closed 2023-02-02 by Ysurac

darthclide, 2023-02-02:
I am confused. I do want to try things. I disabled IPv6 on the eth1 interface (the one I am trying to bypass traffic to). I also want to reset the DNS but without restarting the router so I was asking how to do that so that if this DNS reset is all that is needed, I can quickly do it with 0 reboot in the future when OMR-bypass is acting up.

If you are thinking this is a result of a long router uptime (similar to needing to reboot windows after it has been on for many months) I just updated 4 days ago so I am unsure how OMR-bypass could have messed up in only a few days?

Ysurac, 2023-02-02:
What I was saying is maybe a URL conflict with other or use an incorrect character that make all URLs fail when it's set.

OMR-Bypass, to bypass domains simply use DNSmasq ipset entry: when a request is done for a domain in the list ip is added to ipset. When you try to go to this IP, that is now in ipset, it know what route must be used.
So to have omr-bypass working you need to use only openmptcprouter as DNS and all should always work.

In your case, if you added/deleted domains, config can be messed (should not) and you need to reboot to fix that.

Do you bypass only domains or services too ?

Ysurac, 2023-02-02:
To send the full config of omr-bypass, via ssh: ```uci show omr-bypass```
omr-bypass is run only when config change, it's not a daemon.

darthclide, 2023-02-02:
https://pastebin.com/0c92dbZb

Maybe there is something going wrong when the omr-bypass gets passed to ipset?

I don't think I have ever touched the DNS settings of eth0, but where would I go to verify I am using the stock options for OMR?

darthclide, 2023-02-03:
https://pastebin.com/b2jDZQt0

https://pastebin.com/KA138FCq

darthclide, 2023-02-06:
Some extra info: Using ExpressVPN on my phone fixes the problem. But I guess that makes sense since I am bypassing your whole software stack with that VPN?

darthclide, 2023-02-06:
https://pastebin.com/gMTc8ydR
https://pastebin.com/DRpEkxsS
https://pastebin.com/HpStCq4H

traceroute reddit.com
traceroute to reddit.com (151.101.193.140), 30 hops max, 46 byte packets
 1  10.255.255.1 (10.255.255.1)  11.219 ms  11.353 ms  11.137 ms
 2  *  *  *
 3  104.156.235.33 (104.156.235.33)  12.425 ms  12.155 ms  12.589 ms
 4  *  *  *
 5  *  *  *
 6  *  *  *
 7  *  *  *
 8  *  *  *
 9  *  *  *
10  *  *  *
11  *  *  *
12  *  *  *
13  *  *  *

I stopped it since it seemed to fail.

Ysurac, 2023-02-07:
Strange, as reddit is bypassed it shouldn't use 10.255.255.1 as gateway.
The reddit.com ip, 151.101.193.140, is correctly set in ipset omr_dst_bypass_eth1, iptables set a fwmark 0x                                   5393 and using this via ip rule "from all fwmark 0x5393 lookup 3" it should look at routing table 3 that is:
```
default via 192.168.10.1 dev eth1 table 3
192.168.10.0/24 dev eth1 table 3 scope link
```
So all seems to be correct here.
What is the result of ```ip route get 8.8.8.8 mark 0x5393```, this will check that route configuration is correct.

darthclide, 2023-02-07:
8.8.8.8 via 192.168.10.1 dev eth1 table 3 src 192.168.10.158 mark 0x5393 uid 0
    cache

darthclide, 2023-02-09:
Would restarting my DSL modem do any good? I am open to try anything you suggest.

darthclide, 2023-02-11:
Okay this is odd, I left all other rules enabled, but disabled my port 443 rule and all the url bypass rules work? Is it possible there was an overlap bug where reddit.com and port 443 were both sent to eth1 and it hangs?