Ysurac/openmptcprouter #2752: RUTX50 - OpenMPTCProuter

Tabarane, 2023-01-31:
## Expected Behavior
Download the image and flash it to RUTX50 and it changes to be OPENMPTCP router

## Current Behavior
i tried to use the image proposed from the website that in theory support all RUTX version, but from the default GUI it fails the verification process

## Specifications

RUTX50 5G dual simcard modem

Ysurac, 2023-02-01:
What is the image tried ?
The image is tested on RUTX12, it may work or not on other RUTX.
You can also try to update via bootloader menu: https://wiki.teltonika-networks.com/wikibase/index.php?title=RUTXxx_Firmware_via_The_Bootloader_Menu&mobileaction=toggle_view_desktop

Tabarane, 2023-02-01:
[image]
openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-ipq40xx-generic-teltonika_rutx-squashfs-sysupgrade.bin

Ysurac, 2023-02-01:
Maybe some changes are needed to support RUTX50, I will check.

NKnusperer, 2023-02-13:
@Tabarane any update on this? I'm interested in using this device too.

Tabarane, 2023-02-13:
@NKnusperer im still waiting for an updated firmware from @Ysurac so i can test it on RUTX50

NKnusperer, 2023-02-14:
Just got my hands on this device.
I've seen that the board and device tree was added in c3f3d7134552102e58fa58bb7b6d1b24e972339f , however the nightly image is failing to flash as described above. I noticed that the current SDK `RUTX_R_00.07.03.4` (https://wiki.teltonika-networks.com/view/GPL) contains much more files eg. in `target/linux/ipq40xx/patches-5.4` then what exist here.
Given I have zero knowledge regarding embedded Linux development this might or might not be related to this issue.

abenhamo, 2023-03-12:
Hello, 
any news on this? Planning to purchase the RUTX50 only for this purpose.
thank you

Tabarane, 2023-03-12:
@abenhamo i didn't get any updates for now, so im still waiting

Ysurac, 2023-03-13:
It's added in snapshot build, but no idea if it's working or not.

Tabarane, 2023-03-13:
@Ysurac i can test it, can you share with me the url for the image?

abenhamo, 2023-03-13:
@Tabarane please let us know the outcome

abenhamo, 2023-03-16:
@Tabarane OMG it seems that the rutx50 has one ONE 5g module, while the rutx12 has TWO 4G modules.
so openmptcprouter on rutx50 is completely useless.. at least for cellular aggregation...

Tabarane, 2023-03-16:
@Tabarane i think no need to test then :(( in this case RUTX12 is the best option :(

Tabarane, 2023-03-16:
@abenhamo when i tested the Default 2 simcard failover with their own OS, it didnt not work directly, i had 2 simcard , never saw that both of them in same time got IP address, and when i removed the active simcard , the second one did not startup and it was failure which was my first indication that it looks like 1 modem installed and they aggregate bands only "if there is such thing at all"

abenhamo, 2023-03-16:
Failover needs only one module and two sims. To aggregate you need absolutely 2 seperate modules. RTUX50 is a no-go for mptcp...

giuchie, 2023-05-09:
Some news? i've tried to install the image on https://snapshots.openmptcprouter.com/5.4/rutx/targets/ipq40xx/generic/ but  got a signature error. Even in bootloader state...

alexisries, 2023-07-25:
> Failover needs only one module and two sims. To aggregate you need absolutely 2 seperate modules. RTUX50 is a no-go for mptcp...

It would be interesting to be able to aggregate a 5G connection and a Starlink connection.

closed 2023-10-30 by github-actions[bot]