Ysurac/openmptcprouter #2756: Iperf compilation error  error: redefinition of 'struct in6_flowlabel_req'

fareign, 2023-02-02:
branch master  
OMR_FEED_SRC="v0.58.5" 

> flowlabel.h:41:8: error: redefinition of 'struct in6_flowlabel_req'
 struct in6_flowlabel_req
        ^~~~~~~~~~~~~~~~~
In file included from /root/openmptcprouter/x86_64/5.4/source/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/tcp.h:25,
                 from iperf.h:43,
                 from iperf_tcp.c:41:
/root/openmptcprouter/x86_64/5.4/source/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/in6.h:71:8: note: originally defined here
 struct in6_flowlabel_req {
        ^~~~~~~~~~~~~~~~~
make[6]: *** [Makefile:955: iperf_tcp.lo] Error 1
make[6]: Leaving directory '/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.10.1/src'
make[5]: *** [Makefile:711: all] Error 2
make[5]: Leaving directory '/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.10.1/src'
make[4]: *** [Makefile:389: all-recursive] Error 1
make[4]: Leaving directory '/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.10.1'
make[3]: *** [Makefile:85: /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/iperf-ssl/iperf-3.10.1/.built] Error 2
make[3]: Leaving directory '/root/openmptcprouter/feeds/5.4/packages/net/iperf3'
time: package/feeds/packages/iperf3/ssl/compile#6.36#1.13#9.82
    ERROR: package/feeds/packages/iperf3 failed to build (build variant: ssl).
make[2]: *** [package/Makefile:116: package/feeds/packages/iperf3/compile] Error 1
make[2]: Leaving directory '/root/openmptcprouter/x86_64/5.4/source'
make[1]: *** [package/Makefile:110: /root/openmptcprouter/x86_64/5.4/source/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/openmptcprouter/x86_64/5.4/source'
make: *** [/root/openmptcprouter/x86_64/5.4/source/include/toplevel.mk:230: world] Error 2

Ysurac, 2023-02-02:
Please use template.
Why do you compile an old field ? and did you also use old build scripts ?

fareign, 2023-02-02:
> Please use template. Why do you compile an old field ? and did you also use old build scripts ?

how to use new scripts?

Ysurac, 2023-02-02:
You need to use old build script when you use old feed.

closed 2023-05-08 by github-actions[bot]