Ysurac/openmptcprouter #2758: Video streaming stutters after bonding two (4G SIM) internet.

gaamaaresosa, 2023-02-02:
## Expected Behavior
After aggregating 2 internets video streaming must be smoother.

## Current Behavior
The streamed video not smooth.
Stuttering as if not enough bandwidth.

We try video streaming from local PC via OBS to a Video Streaming server with this bonded internet.
The MPTCP bandwidth window shows each internet has enough (5 to 6 Mbps) bandwidth.
But my doubt is the aggregation is happening or not.
(How to confirm the Bonding is successfully happneing?)

If I disable OMR and directly streaming from any single internet, the video is smoother.

Our bonding and aggregation logics are taking care of video streaming standard right ?

I tried 9Mbps, 8 Mbps 6 Mbps and even 5Mbps bandwidth on my video streaming software.

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 
  - OpenMPTCProuter VPS version: 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: DigitalOcean
  - OpenMPTCProuter platform: RPI4 Kernal 5.4.194

STATUS:
[image]

BANDWIDTH:
[image]

SPEED TEST OF VPS FROM CLIENT: (omr-iperf vps -R)
[image]

SPEED TEST ON VPS:
[image]

WAN1:
[image]

WAN2:
[image]

Ysurac, 2023-02-02:
Did you se download and upload speed for each interface ? if yes, set lower values

gaamaaresosa, 2023-02-02:
I am not sure how to see upload speed per interface.
I tried 
omr-test-speed eth1
omr-test-speed eth2

But that shows only download speed only.
https://github.com/Ysurac/openmptcprouter/discussions/2757

[image]

Ysurac, 2023-02-02:
You should know download/upload your ISP give you. Else you may use iperf for exemple.

ccmks, 2023-02-03:
Did you use V2RAY for aggregation or Shadowsocks? I think for video streaming, V2RAY might be better

gaamaaresosa, 2023-02-03:
> V2RAY

Could you please let me know where that option is as I am new to OMR ?

gaamaaresosa, 2023-02-03:
@ccmks
Yes, by default it's in V2RAY only.
[image]

gaamaaresosa, 2023-02-03:
I strongly doubt the bonding/aggregation are happening or not, while I do video streaming with 2 internets.
5Mbps + 5Mbps = I get only 5Mbps.

Where can I see the actual bodning graph ?
Why this bond0 and bonding_master graph always blank while I consume internets ?
Am I missing something on my setup ?

[image]

ccmks, 2023-02-03:
You need to go to advanced settings and select default proxy to V2RAY

[image]

gaamaaresosa, 2023-02-03:
Btw I hope you too use the latest version of openmptcprouter v0.59.1-5.4.
[Powered by openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c](https://github.com/ysurac/openmptcprouter)

I see no such option in "Advance Settings"
I use openmptcprouter and no OpenWrt

gaamaaresosa, 2023-02-03:
Thanks ccmks,
Your v2Ray trick seems working fine.
I will test more...

ccmks, 2023-02-03:
V2RAY supports UDP aggregation. It is much better compare to shadowsocks when it comes to streaming

gaamaaresosa, 2023-02-03:
> V2RAY supports UDP aggregation. It is much better compare to shadowsocks when it comes to streaming

But I do rtmp:// streaming where udp is not used.
Still its better ?

gaamaaresosa, 2023-02-03:
Btw please help me to fix this issue also.
When I test each internet (SIM card usb dongle Huawei) I get 2 Mbps
But both together, only one is 3Mbps and the other one always 1 Mbps.
Please see this image.
I doubted the cellular provider, but when I check individually both are same.

[image]

ccmks, 2023-02-03:
There is a chance that your VPS doesn't have good connection when it comes to streaming. What happened if you just use only 1 connection in your OMR, is the streaming still good or not?

gaamaaresosa, 2023-02-03:
> There is a chance that your VPS doesn't have good connection when it comes to streaming. What happened if you just use only 1 connection in your OMR, is the streaming still good or not?

Yes with single internet always very smooth. But we need to set the bandwidth lesser as per single net speed.

ccmks, 2023-02-03:
The speedtest doesn't tell about streaming, you actually have to try with streaming directly from your VPS

I have seen some VPS whose internet is good for speedtest, but sucks for other service

gaamaaresosa, 2023-02-03:
> The speedtest doesn't tell about streaming, you actually have to try with streaming directly from your VPS
> 
> I have seen some VPS whose internet is good for speedtest, but sucks for other service

Oh !

ccmks, 2023-02-03:
You need to test your 2nd LTE connection to your VPS to see if you are getting the full speed. Some ISP might throttle your internet connection to your VPS, but not to speedtest.

This is very common practice ISP in my country

gaamaaresosa, 2023-02-03:
Is there an option to confirm the bonding and aggregation is working fine on our OMR as per the settings ?
Like a graph ?
Is there any built in speed test inside OMD ui itself ?

gaamaaresosa, 2023-02-03:
> You need to test your 2nd LTE connection to your VPS to see if you are getting the full speed. Some ISP might throttle your internet connection to your VPS, but not to speedtest.
> 
> This is very common practice ISP in my country

In my case both SIM cards are by same ISP provider, I got at same at time.

ccmks, 2023-02-03:
> In my case both SIM cards are by same ISP provider, I got at same at time.

This even worse. Same ISP, same Cell tower, 2 SIMs ended up fighting each other for over single bandwidth on single cell tower

gaamaaresosa, 2023-02-03:
> > In my case both SIM cards are by same ISP provider, I got at same at time.
> 
> This even worse. Same ISP, same Cell tower, 2 SIMs ended up fighting each other for over single bandwidth on single cell tower

You are right !
Good chances are there. I will think...

gaamaaresosa, 2023-02-03:
> > In my case both SIM cards are by same ISP provider, I got at same at time.
> 
> This even worse. Same ISP, same Cell tower, 2 SIMs ended up fighting each other for over single bandwidth on single cell tower

You don't doubt we set one interface as "Multipath TCP = Master and other Enabled"

Ysurac, 2023-02-03:
Multipath Master or enabled doesn't change speed.
Master is used for initial connection, then connections are used based on latency.

gaamaaresosa, 2023-02-03:
> Multipath Master or enabled doesn't change speed. Master is used for initial connection, then connections are used based on latency.

Thanks Ysurac,
I think the Huawei dongle is very deligate.
If I use USB3 hub I get poor speed, USB2 link cable increase speed.
How to workout 4 Nos 4G SIM LTE inside 1 Raspberry Pi 4 OMR I don't know.
May be SIM 4G LTE module to be tried.

gaamaaresosa, 2023-02-03:
@Ysurac
One question please.
I see on the status, when we don't consume bandwidth, the VPS server is active.
In fact all OMR and local interface are active.
After start consuming bandwidth, the VPS server says "can't contact server admin script" and goes off.
This may be logical as the OMR could not hand shake with VPS over internet or vice versa.
But practically everone will face this situation and process will fail right ?
Or we should not consume full available band width ?

Ysurac, 2023-02-03:
In the wizard, you need to set SQM with upload and download speed, and enable SQM autorate for unstable connections like 4g, satellite,...

ccmks, 2023-02-03:
> Thanks Ysurac, I think the Huawei dongle is very deligate. If I use USB3 hub I get poor speed, USB2 link cable increase speed. How to workout 4 Nos 4G SIM LTE inside 1 Raspberry Pi 4 OMR I don't know. May be SIM 4G LTE module to be tried.

From my experience here in my country, the USB LTE dongle tend to be unstable for long run. Depend on the type and brand. In my case, I always go with CPE LTE modem that it has RJ45 LAN port to connect to OMR router. That has always been stable for me, YMMV!

gaamaaresosa, 2023-02-03:
> From my experience here in my country, the USB LTE dongle tend to be unstable for long run. Depend on the type and brand. In my case, I always go with CPE LTE modem that it has RJ45 LAN port to connect to OMR router. That has always been stable for me, YMMV!

Thanks agan,
I will consider it. Infact I need the entire stuff as portable for outdoor location live streaming solution like LiveU Solo. 
Speedify has a complete software solution + VPN on the cloud.
That software client we can install on any Windows/Linux?Mac/Android systems without special hardwar.
But similar modems needed.
Now in our case,
Raspberry Pi (SD Card not stable and corrupts often) with battary power backup
USB hub with battary power backup
4 dongles or modems for SIM LTE
And a laptop for video streaming.

ccmks, 2023-02-03:
This is my similar setup too that I have been thinking for building it

In fact that some CPE LTE modem has battery built-in, some of them just using regular 12V DC. The think that saved me is I bought USB to 12V DC connector, so I can just plug in my CPE LTE modem to power bank and it will become portable. Just to make sure that your powerbank has 4 different USB port so you can plug in each port to your network equipment.

I used to use speedify and I didn't like the overall performance. In fact, it even slower when bonding compare to single connection due to the server is always congestion (what do you expect on public server), whereas OMR, we have our own VPS that the resource is much dedicated to us and no one else is using our server.

gaamaaresosa, 2023-02-03:
@ccmks
> I used to use speedify and I didn't like the overall performance. In fact, it even slower when bonding compare to single connection due to the server is always congestion (what do you expect on [...]

Exactly same bad experiance for me too on Speedify.
Annoying GUI keeps on poping up messages and alerts.
Our own VPN is always better.
Have you succeeded in Raspberry Pi 4 + OMR combo for live streaming ?
If yes please give me some tips for do and don'ts

gaamaaresosa, 2023-02-03:
> In the wizard, you need to set SQM with upload and download speed, and enable SQM autorate for unstable connections like 4g, satellite,...

Thanks again Ysurac,
I will give a try soon.

> In the wizard, you need to set SQM with upload and download speed, and enable SQM autorate for unstable connections like 4g, satellite,...

Can you please help me what values to set on upload and download speed.
What is the formula ?
And relevent video link ?

ccmks, 2023-02-03:
I have never used Raspberry Pi 4, but I used banana-pi (forgot the model, but it is one of the supported in OMR device list) before and they are running great, it has 5 gigabit ethernet port. The only thing I don't like is that it won't do auto boot after power loss, there is workaround to modify the board, but I didn't bother to try it. I think the newer model don't have this problem, but need to have @Ysurac to clarify if it is working with OMR or not.

I can't recommend any modem as the manufacture and government regulation is different from country to another country.

I live in Indonesia where the 4G LTE coverage is not spread on all area, and it has lots of island in one country. There are still lots of blind spot. However, I joined to the facebook group that is all about the GSM community and people are struggling to get LTE signal in remote places. That's where I learnt a lot from there.

The only thing I can give you just general idea on the setup, such as:

- Portable CPE LTE modem (some of them has built-in battery, some of them can be plugged in to micro-USB)
- USB A to 12V DC connector
- Banana-Pi or similar device
- Mini-router like TP-Link MR3020 or similar for wifi AP

You have to do lots of research on that before buying.

My current project, I am currently focus for OMR just to be stable enough to handle small business environment as the internet dedicated replacement. But I am still struggling the find the sweet-spot VPS that can serve proper internet connection. Most VPS here got throttle from ISP due to monopoly and capitalism.

Hope this helps

gaamaaresosa, 2023-02-03:
@ccmks
Very nice explanation, thanks.
I agree all your points.
I just dived in on OMR.
More than OMR architecture, I love **Ysurac's** fire on this project.
Very prompt and swift reponse on queries.
That's the only reason made me to spend useful time on this project.
Thanks to **Ysurac** !

ccmks, 2023-02-03:
This OMR solution is like God-gift from @Ysurac 

This will solve lots of internet issue here that I am facing in my country! Definitely need this project kept going on

gaamaaresosa, 2023-02-03:
> In the wizard, you need to set SQM with upload and download speed, and enable SQM autorate for unstable connections like 4g, satellite,...

When I enable SQM the speed goes down.
(Download/Upload = 50000,  Enable SQM autorate = Ticked)

[image]

Ysurac, 2023-02-03:
SQM limit speed to limit congestion, in your result are you sure you set 50000 and not 5000 ?

gaamaaresosa, 2023-02-03:
> SQM limit speed to limit congestion, in your result are you sure you set 50000 and not 5000 ?

Sure I set 50000 (50Mbps) also tried 15000 (15Mbps)
First of all my both SIM cards confusing me as keep on showing fluctuated speeds.
I bother about only upload speed as I do video streaming.
So some time it show 20Mbps upload and some time 5Mbps and some time even 1Mbps.
The Cellular provider is famous here.
When I do same SIM with mobile tethering I get never less than 15Mbps upload.
So I can't doubt the SIM
I have TP-Link TL-MR3020 (Useless because I need to insert same Huawei USB dongle in that) 
I have to look for some better option to connect SIM 4G LTE internet like mobile tethering.

gaamaaresosa, 2023-02-03:
@Ysurac
Can you please some how clear my doubt as I 90% suspect only one interface sharing its internet for my streaming.
Though I see both interface show bandwidth graph on the Network > MPTCP > Bandwith tab.
But I get only single interface speed.

Is there any way I can make sure I see 100% aggregation is happening.
The upload graph shows each interface upload speed 4 to 5 Mbps
That mean I should use 8 to 10Mbps speed and stream.
But I could set only 4Mbps steam speed for smooth streaming, else video stutter.

This is what below command also shows (4.99 Mbits/sec)
Which is correct ? How to get 8 to 10Mbps speed and stream.

omr-iperf vps
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  5.95 MBytes  4.99 Mbits/sec    0             sender
[  5]   0.00-11.19  sec  5.11 MBytes  3.83 Mbits/sec                  receiver

gaamaaresosa, 2023-02-03:
Kindly compare these images.
I did video streaming with ~3Mbps speed.
Individual interface bandwidth graph was set side by side and monitored.
In which first interface doesn't use its internet.
But 2nd interface use its assigned internet properly.

[image]

Below 2nd image was captured at the same time.
Where the upload bandwidth was monitored for both the interfaces simultaneously.
In which it shows both interfaces are using their internet equally.

Practically I see from day one, only single interface internet was used properly.
Please correct me if I am wrong in monitoring.
 
[image]

Ysurac, 2023-02-03:
As it's USB modems, unplug the modem with highest speed and check the result.
Also check that all is green in status page when you do upload test.

gaamaaresosa, 2023-02-03:
> As it's USB modems, unplug the modem with highest speed and check the result. Also check that all is green in status page when you do upload test.

I had confirmed all are green in the status page.
Further test I will do on Monday and update.
Thank you so much.

gaamaaresosa, 2023-02-06:
@Ysurac
I am back to my desk and spending several hours to make my 4G LTE Huawei dongles to get stable internet first. I use powerd USB3 8 port hub and connecting 2 nos of 4G Huawei dongles and that hub I connect on Raspbery Pi 4 OMR.

Initially all are green on Network overview.
As I started consuming the internet (I do speed test with omr-iperf vps) all are start showing warning. 
But the speed test gives result first time and then failes at 2nd attempt as the warning appears.

I know this issue is point many times by many people.
Logically the VPS server handshaking happening with OMR client's internet.
But when all internets are consumed by SpeedTest, the VPS server ping fails and started loosing the connection and gives warning, so the OMR clients SpeedTest also fails.

1) What is the workaround to manage this issue ?
I tried SQM but its is giving worse result and internet becomes still slower.

2) Is there any ping (Watchdog) time out adjustment to the VPS server for 4G LTE unstable/poor latency internets ?

3) Please let me know where to learn how to add additional 4G LTE dongles/LTE to LAN modems on OMR successfully. Any step by step video tutorial ?

1ST TIME:
-----------
root@OpenMPTCProuter:~# omr-iperf vps
Connecting to host  XX.XX.XX.199, port 65400
[  5] local 192.168.10.12 port 54039 connected to  XX.XX.XX.199 port 65400
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   134 KBytes  1.09 Mbits/sec    2   12.6 KBytes
[  5]   1.00-2.00   sec   103 KBytes   843 Kbits/sec    2   12.6 KBytes
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    0   12.6 KBytes
[  5]   3.00-4.00   sec   149 KBytes  1.22 Mbits/sec    0   12.6 KBytes
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec    0   12.6 KBytes
[  5]   5.00-6.00   sec   189 KBytes  1.55 Mbits/sec    0   12.6 KBytes
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    0   12.6 KBytes
[  5]   7.00-8.00   sec   253 KBytes  2.07 Mbits/sec    0   12.6 KBytes
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0   12.6 KBytes
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    0   12.6 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   827 KBytes   678 Kbits/sec    4             sender
[  5]   0.00-12.67  sec   401 KBytes   259 Kbits/sec                  receiver

2ND TIME:
-----------
root@OpenMPTCProuter:~# omr-iperf vps
Connecting to host XX.XX.XX.199, port 65400
iperf3: error - test authorization failed

[image]

[image]

gaamaaresosa, 2023-02-06:
I have pulled out all my USB hub and Huawei dongles.
Now directly I have connected a Mobile with same SIM card and tethered on Raspberry Pi 4 USB3 port.

Kindly check all these images.
Very first all green, then only by one randomly goes off with some warning.
Even if I consume the internet for speedtest, these warning I get.
After that the green comes and goes by its own automatically.
If short my OMR is unstable.

Even I have confirmed Shadowsocks on proxy.

Where is the issue on my OMR ?
Same SIM card on the mobile shows ~22Mbps download and ~16Mbps upload.
Where on OMR it hardly shows ~2Mbps download and ~1.5Mbps upload

[image]

[image]

[image]

[image]

[image]

[image]

EhsanVahab, 2024-01-03:
dear @gaamaaresosa have you ever solved your problem?
I get stuck in the very problem you got

gaamaaresosa, 2024-01-03:
Unfortunately I couldn't solve it and I have dropped this project.

⁣gaamaa​

EhsanVahab, 2024-01-05:
it seems openmptcprouter is not still a perfect solution. so need more time

ccmks, 2024-01-06:
Have you tried to use speedify? I heard they are bragging about the video
streaming through bonding

he852100, 2024-01-08:
两张卡带宽相等的情况下，最大带宽只有总带宽的70%，其它情况带宽按最小的那路翻倍。信号带宽波动可能导致断流，所以这玩意儿并不适合户外移动直播