Ysurac/openmptcprouter #2759: "invalid sector size 65535" error on router boot after install

ken-sheppard, 2023-02-03:
## Expected Behavior

upon boot openmptcprouter starts

## Current Behavior

- after imaging the disk and booting I get "invalid sector size 65535"
- after I press enter I get the boot menu and both OpenMPTCProuter and OpenMPTCProuter(failsafe) generate the same error

## Possible Solution

I assume I am doing something wrong, but something may be wrong with the images. I can install and boot OpenWrt latest x86-64 ext4 efi with no issues.

## Steps to Reproduce the Problem

  1. unzip and copy *.img to Ubuntu 22.04.01 LTS USB install drive
  2. boot USB drive on x86-64 machine
  3. Try Ubuntu boot option
  4. Try Ubunto desktop option
  5. At a terminal: 
  `sudo dd bs=1M if=openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined-efi.img of=/dev/nvme0n1 conv=fsync
576+1 records in
576+1 records out
604274176 bytes (604 MB, 576 MiB) copied, 5.32785 s, 113 MB/s
  7. OR disks, restore image and select img file
  8. Reboot

I tried on two different drives(/dev/nvme0n1 and /dev/sda), one a time, with several variations of the dd command and with gnome-disks

Same issue with ext4 and sqashfs images

ext4 image (EFI): 
openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined-efi.img.gz
AND
squashfs image (EFI): openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-squashfs-combined-efi.img.gz

## Context (Environment)

- new Celeron mini PC with 500GB NVME and 64GB SATA drives
- I can get latest OpenWrt and HomeAssistent to image and boot using dd

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4-r0+16594n) -->
  - OpenMPTCProuter VPS provider: NA
  - OpenMPTCProuter platform: x86_64) 
  - Country: USA

Ysurac, 2023-02-03:
Your host only support UEFI ?

ken-sheppard, 2023-02-03:
Yes, it is a new machine that only supports UEFI.

ken-sheppard, 2023-02-05:
This build boots as expected:

https://snapshots.openmptcprouter.com/6.1/x86_64/targets/x86/64/openmptcprouter-v0.59.2alpha-6.1-r0+21757-895f38ca1e-x86-64-generic-ext4-combined-efi.img.gz

closed 2023-05-11 by github-actions[bot]

Thormus, 2024-03-08:
What's the fix for this? I'm stuck here.