Ysurac/openmptcprouter #2763: need help in adding 4 usb 4G dongles

rajeshkrarora1, 2023-02-05:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

rajeshkrarora1, 2023-02-05:
created a usb0 interface, and was able to add 1st usb 4G dongle, for second dongle unable to create usb1 interface, please suggest how can I add and configure second, third and fourth usb 4g dongles on raspberry pi 4 8gb 
openmptcprouter version v0.59.1 rpi4 aarch64_cortex-a72

rajeshkrarora1, 2023-02-05:
Openmptcprouter version v0.59.1 rpi4 aarch64_cortex-a72

OpenMPTCProuter VPS version:        Linux vps-51f22e61 5.4.207-mptcp

OpenMPTCProuter VPS provider:  OVH

OpenMPTCProuter platform: rpi4

Ysurac, 2023-02-05:
You can do what you want in Network->Interfaces.
But 4 dongle on RPI4 without a powered USB hub, I think this will not work correctly.

rajeshkrarora1, 2023-02-06:
Thanks for your prompt reply, I have purchased a powered 8-port Network switch  TL-SG108E, though I have connected two usb 4G router dongles directly to raspberry pi usb ports I was able to configure them to work as usb0 and usb1 devices, but when I tried to configure a third device it is not working(maybe due to lack of power). I do not know how to configure them through the switch, the guide I am using for network switch is https://kamrul.dev/aggregate-multiple-internet-with-openmptcprouter/ , but when is go to System>OpenMPTCProuter>Status. the Network overview shows me "no data"

gaamaaresosa, 2023-02-06:
@rajeshkrarora1
I think when you connect 4G dongle (USB) on powered hub, you should see eth1, eth2...
Instead of usb0, usb1 ...
Please confirm that with "ip addr" command.
If down then up it with ifconfig up eth1

rajeshkrarora1, 2023-02-07:
Thanks for your guidance, can attach up to 8 usb 4G/5G dongles this way?

gaamaaresosa, 2023-02-07:
> Thanks for your guidance, can attach up to 8 usb 4G/5G dongles this way?
Practically 4 lte usb dongles only can be connected on 8 port usb hub as the dongle size is bigger than usb pendrives. Other wise you must test it 8 dongles by your own.

rajeshkrarora1, 2023-02-08:
@gaamaaresosa Thanks for your prompt response🙏🙏🙏🙏issue resolved

closed 2023-02-08 by rajeshkrarora1